PowerBroker2/ELMduino #255: setPIN error when using ESP32

JayKapse, 2024-07-02:
I'm trying to connect to ELM327 adapter using the ESP32_test code. When i run it without uncommenting ELM_PORT.setPin("1234"); it does not connect to the adapter. When I include the line in the code, i get this error:

/private/var/folders/fm/2sf3tvf92vxcwskbwrwpjpyr0000gn/T/.arduinoIDE-unsaved202463-50295-11f7gil.lh47/ESP32_test/ESP32_test.ino: In function 'void setup()':
/private/var/folders/fm/2sf3tvf92vxcwskbwrwpjpyr0000gn/T/.arduinoIDE-unsaved202463-50295-11f7gil.lh47/ESP32_test/ESP32_test.ino:27:12: error: 'class BluetoothSerial' has no member named 'setPin'
   27 |   ELM_PORT.setPin("1234");
      |            ^~~~~~

exit status 1

Compilation error: 'class BluetoothSerial' has no member named 'setPin'

I'm not sure why this is, I've tried installing BluetoothSerial library and that gives a whole new set of errors. Relatively new to this environment, any help would be appreciated. Thank you! 

Here's the code:

`
        
        #include "BluetoothSerial.h"
        BluetoothSerial SerialBT;
        
        
        #define DEBUG_PORT Serial
        #define ELM_PORT   SerialBT
        
        
        void setup()
        {
          //pinMode(LED_BUILTIN, OUTPUT);
          //digitalWrite(LED_BUILTIN, HIGH);
        
          DEBUG_PORT.begin(115200);
          ELM_PORT.begin("ESP32test", true);
          ELM_PORT.setPin("1234");
        
          DEBUG_PORT.println("Attempting to connect to ELM327...");
        
          if (!ELM_PORT.connect("OBDII"))
          {
            DEBUG_PORT.println("Couldn't connect to OBD scanner");
            while(1);
          }
        
          DEBUG_PORT.println("Connected to ELM327");
          DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
          DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
          DEBUG_PORT.println();
        }
        
        
        void loop()
        {
          if(DEBUG_PORT.available())
          {
            char c = DEBUG_PORT.read();
        
            DEBUG_PORT.write(c);
            ELM_PORT.write(c);
          }
        
          if(ELM_PORT.available())
          {
            char c = ELM_PORT.read();
        
            if(c == '>')
              DEBUG_PORT.println();
        
            DEBUG_PORT.write(c);
          }
        }
`

PowerBroker2, 2024-07-03:
No idea, you should ask in the Espressif and/or Arduino forums perhaps

jesusangel72, 2024-07-04:
it seems like espressif defined CONFIG_BT_SSP_ENABLED in compiled sdk for Arduino IDE, so you can't modify it or set some sdkconfig.ini parameter, in BluetoothSerial.h you can find:

`#ifdef CONFIG_BT_SSP_ENABLED
  void enableSSP();
  void enableSSP(bool inputCapability, bool outputCapability);
  void disableSSP();
#else
  bool setPin(const char *pin, uint8_t pin_code_len);
#endif`
So setPIN function is not defined.
You can try to modify BluetoothSerial.h and move setPIN out of #else and dissble SSP before begin function as mentioned in comments.
This code seems to be update in version 3.0.x of esp32 package you can try 2.0.x that I think is the one used by PowerBroke2r when developing ELMDuino.

I'm facing the same problem, so I will try both ways too.

Regards

> I'm trying to connect to ELM327 adapter using the ESP32_test code. When i run it without uncommenting ELM_PORT.setPin("1234"); it does not connect to the adapter. When I include the line in the [...]

JayKapse, 2024-07-04:
Thank you @jesusangel72 and @PowerBroker2 . I downgraded to 2.0.17, and it works like a charm!

closed 2024-07-04 by JayKapse