PowerBroker2/ELMduino #258: BT simple connection failed

gonzob, 2024-08-04:
I've had a ESP32 device monitoring my MG ZS EV for over six months. Here's what I built:

https://github.com/PowerBroker2/ELMduino/issues/202

Then yesterday it stopped working. I tested various aspects of the system trying to find the problem and I eventually discovered that the simple call to connect wasn't working anymore. It was this statement:

`if (!ELM_PORT.connect(bt_name))`

I tried connecting using this simple command but with the mac address, but that failed also.
 
I then checked with the Arduino BluetoothSerial examples and was able to get the board to connect using the DiscoverConnect program. 
I then used the same method as that program used - it is a more complex connection command:

`if (!ELM_PORT.connect(bt_mac,  channel, sec_mask, role))`

This worked. I'm thinking that for some reason the channel number changed in the bluetooth dongle???

The device is working now. Has anyone experienced this before?

Gonzo

wcjxixi, 2024-08-09:
You might try to remove paired devices using [this sketch](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/bt_remove_paired_devices/bt_remove_paired_devices.ino).

gonzob, 2024-08-09:
Tried that, and it had no effect.

Turksat46, 2024-09-24:
There is more to that. 
First of all, connecting with device name does take longer and does make more errors.
Try connecting to your device with the mac address like the following:
`
//save mac address as uint8_t
uint8_t address[6]  = {0xDC, 0x0D, 0x30, 0xC8, 0x4F, 0x0A};
//connect with mac address
  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to Bluetooth OBD scanner");
    //while(1);
  }
`

Also, use an older version of the esp32 library, I use 2.0.14 in which the usePin() method still works.

closed 2025-02-09 by PowerBroker2