PowerBroker2/ELMduino #260: Arduino Nano, How to reduce Connection Time

MFAISALREHMAN, 2024-09-24:
Hello
My My arduino nano connects to ELM327 in about 20 seconds. Is there anyway to reduce this time e.g knowing the connection protocol name and calling the initialize function. Thankx.

I am getting following msgs.

 �Attempting to connect to ELM327...
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: A
	Received char: T
	Received char: _
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
Received char: _
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
Received char: A
	Received char: T
	Received char: _
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT SP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: 0100
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: B
	Received char: E
	Received char: 3
	Received char: E
	Received char: B
	Received char: 0
	Received char: 0
	Received char: 0
Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 4100BE3EB000
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: A
	Received char: E
	Received char: B
	Received char: \r
	Received char: \r
	Received char: >
elimiter found.
All chars received: 410C0AEB
Expected response header: 410C
Single response detected
64-bit response: 
	responseByte_0: 235
	responseByte_1: 10
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
responseByte_7: 0
rpm: 698.75  Service: 1
PID: 5
Normal length query detected
Query string: 01051
Clearing input serial buffer
Sending the following command/query: 01051

PowerBroker2, 2024-09-24:
It looks like it's working, not sure why the connection would start slow

jimwhitelaw, 2024-09-26:
@MFAISALREHMAN Are you connecting via Bluetooth or a direct serial connection? If it is BT, you might find it establishes a connection faster if you connect with the device address instead of by name.

closed 2025-02-09 by PowerBroker2