PowerBroker2/ELMduino #262: ELM ESP32 suddenly stopped working after Torque used...

miguelos6, 2024-10-10:
Let me try to be as brief as possible.
I successfully used ELMduino old version (I believe 2.x) from 2020... and that worked for 4 years straight...
Then suddenly after I tried to clear some errors using Torque (I couldn't btw) - it no longer works...

I thought it changed something on OBD module (vgate icar pro BT3.0 version, BT name = V-LINK), so I tried to upgrade firmware or factory reset - to no avail..
Then I modified my sketch to latest version, and I believe it's correct and uploaded it to very similar ESP32 module - still does not work.
I got other 2 BT modules, named OBDII and ANCEL, modified sketches accordingly.. still no go..
I tried to removed pair devices sketch (still I don't know it does not matter on new module..) - still no go..

What is weird - android apps all work properly using above 3 modules, at least connect and get some basic values (like rpm), but ESP (2 different modules) do not connect at all to all 3 of these..

basic sketches like ESP32_test.ino and ESP32_Bluetooth_Serial.ino do not connect at all ...
could Torque have changed something on baud speed or sth ? I guess it's the only possibility..

Ahead of me there are still other 2 tests : 
- connect using MAC instead of name
- change speed for BT serial (I need example, couldn't find one)

any help on this matter is highly appreciated...

miguelos6, 2024-10-18:
so basically I modified my old sketch for latest ELMduino 3.x and it didn't work.
Used latest 2.6.4 Elmduino and it works again..

Now weird thing is .. I'm testing now with 2 ESP modules. 
One works with flawlessly, second asks for PIN.
both of these are identical... but for one I get :

non working : 
19:36:07.647 -> [  5778][I][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT: spp connect to remote 00:1d:a5:20:bc:21 channel 1
19:36:07.647 -> [  5792][I][BluetoothSerial.cpp:334] esp_spp_cb(): ESP_SPP_CL_INIT_EVT handle:130 sec_id:56
19:36:07.679 -> [  5813][I][BluetoothSerial.cpp:532] esp_bt_gap_cb(): ESP_BT_GAP_PIN_REQ_EVT (min_16_digit=0)
19:36:07.679 -> [  5822][I][BluetoothSerial.cpp:537] esp_bt_gap_cb(): Input pin code: ""=0x0
19:36:07.679 -> [  5833][E][BluetoothSerial.cpp:525] esp_bt_gap_cb(): authentication failed, status:9
19:36:07.712 -> [  5841][I][BluetoothSerial.cpp:310] esp_spp_cb(): ESP_SPP_CLOSE_EVT status:0 handle:130 close_by_remote:1 attempt 0
19:36:07.712 -> [  5852][D][BluetoothSerial.cpp:790] waitForConnect(): connection closed!

for working : 
19:19:49.712 -> [  5118][I][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT: spp connect to remote 00:1d:a5:20:bc:21 channel 1
19:19:49.744 -> [  5137][I][BluetoothSerial.cpp:334] esp_spp_cb(): ESP_SPP_CL_INIT_EVT handle:130 sec_id:56
19:19:49.880 -> [  5282][I][BluetoothSerial.cpp:296] esp_spp_cb(): ESP_SPP_OPEN_EVT
19:19:49.880 -> Clearing input serial buffer

DerKleinePunk, 2024-10-19:
I think that ist not an Problem with this Lib it ist an other ESP Version loaded / used during Compiling.

https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT_SSP/SerialToSerialBT_SSP.ino

SerialBT.confirmReply(true);

Mabe Help

miguelos6, 2024-10-19:
ok, so it's no longer connected with original thread, but maybe we can continue..

So, basically there is no difference between these 2 boards. The look identical, report itself almost identical in "Get board info" - only SN differs. I'm compling/uploading same sketch, same parameters. And while connecting with 2 different OBD2 - one is always asking for PIN, second connects without this...

DerKleinePunk, 2024-10-19:
Stupi Question you have tried to set an Pin ?

The Chips are Build on same Time ?

Sorry not realy an Idea why one fire the even and the other not.
Some one has set the default Passwort 000 in Hardware...

miguelos6, 2024-10-19:
I cannot set pin as Bluetoothserial uses SPP now. Any attempt to use ELM_PORT.setPin(pin) fails at compilation..

I would need to recompile a lot :
https://github.com/espressif/arduino-esp32/issues/4566#issuecomment-1653772142

DerKleinePunk, 2024-10-22:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT_SSP/SerialToSerialBT_SSP.ino

You can use the Callback for Pin but this ist not Problem of this Lib.  Im Out on this Point.

Erictric, 2024-10-23:
I had a similar issue. I was reading custom PIDs from a Hyundai Kona Electric for the last 4 years. (My program was based on a program written by CiberElectronic in this post: [https://github.com/PowerBroker2/ELMduino/issues/44] )
Last year, I updated the ELMduino library and the program stopped working. I could not figure out how to get my program working with the latest library. I end up reverting to version 2.41 and problem was fixed.
Last month, I changed the Kona for a Ioniq 5. Most PID are the same between the 2 cars. My program was not working on the Ioniq 5. After some troubleshooting, I found that the BluetoothSerial debug messages in the Serial monitor were corrupting the payload received from the ELM327 that I pass to a SafeString fonction. I was getting this message:

Error: createSafeStringFromCharPtrWithSize( data, ...) passed unterminated buffer of strlen >=145

I added DEBUG_PORT.setDebugOutput(false); in the setup() and it fixed the issue.

miguelos6, 2024-10-28:
so I have 3 similar (if not the same) LilyGo esp32 modules now..
1st - the one that worked for 4 years, then suddenly stopped
2nd - worked after downgrading Elmduino to 2.6.4
3rd - not working, giving "authentication failed, status:9" after "Input pin code: ""=0x0"

so I decided to re-flash 1st with common code (working for 2nd), as it was not connecting to my module anyway.. and to my surprise it got the same error as 3rd... I really wonder where the issue is - could OBD map them somehow ? so 1st, and 3rd are not usable with my OBD now...

miguelos6, 2024-10-31:
With additional debugging, comparing line by line I found this thread... 
https://github.com/espressif/arduino-esp32/issues/8153
downgraded boards to 2.0.17 and everything worked...

closed 2024-11-01 by miguelos6