PowerBroker2/ELMduino #263: help on code conversion from ELMduino 2.x to 3.x

miguelos6, 2024-11-01:
Hi folks

For long time I used successfuly the ELMduino 2.x code, with custom PIDs (suggested by PowerBroker2) as below :

```
  if (myELM327.queryPID(34, 4906))
  {
    int16_t tempLitersLeft = myELM327.findResponse() / 64.0;    
    if (myELM327.status == ELM_SUCCESS) { litersLeft = tempLitersLeft; }
    else printError();
  }
```

I tried to understand and convert the same to ELMduino 3.x but the results are gibberish (not the same as on ELMduino 2.x)

my approach 
```
  float tempLitersLeft = myELM327.processPID(34, 4906, 1, 1);
  if (myELM327.nb_rx_state == ELM_SUCCESS) { litersLeft = (int16_t)tempLitersLeft; }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
```

Can someone help me on this ?
I'd like to test new features like DTC from 3.x...

PowerBroker2, 2024-11-01:
Can you post the entire sketch plus the debug output? Also note that the examples have been updated along with the codebase to help the 2.x to 3.x conversion

miguelos6, 2024-11-01:
I don't have debug yet, I was dumping the results on screen only. 

I didn't manage to run any ELM emulator successfully :/
ELMulator from https://github.com/jimwhitelaw/ELMulator throws an error for everything while connected with android app

I just thought there is some major flaw in my understanding of above code...

miguelos6, 2024-11-02:
the sketch is pretty much modified version of what we discussed years ago :)
https://github.com/PowerBroker2/ELMduino/issues/4#issuecomment-593643276

Will try and get the dump tomorrow...

miguelos6, 2024-11-03:
debug from ESP module :
[https://pastebin.com/raw/LEUAK1Qi](url)

jimwhitelaw, 2024-12-04:
I think I may know what's happening there - the number of payload characters is being miscalculated in the response. In previous examples of custom PID queries, the ECU would add extra padding in the response, but yours is not doing that so then the number of expected payload chars is incorrect and you're getting the warning message and 0 value. I've assigned the issue to myself and I'll work on a fix.

miguelos6, 2024-12-05:
is there any way I can help ? 
just to clarify - same module, but with latest ELMduino 2.x works fine. Snippets of code are in first post..

PowerBroker2, 2024-12-06:
Please post the entire sketch **(in tags!)** you're running so that we can better make sense of what's happening in the debug log

miguelos6, 2024-12-06:
with removed unneccessary  display code..

```

#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>
#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_PORT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32-M1" // module name here
#define READDELAY 500
// V-LINK real => 00:1D:A5:20:BC:21
  uint8_t btaddress[6]  = {0x30, 0x3D, 0x35, 0x30, 0x3C, 0x31};

#define OBD_NAME "V-LINK"

// ---------- ENABLE TESTING HERE -------- //
bool testing = false;

BluetoothSerial SerialBT;
ELM327 myELM327;

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

  int16_t rpm=-9, soot=-8, burn=-7, km=-6, litersLeft=-5;
  
void setup() {

  delay(500);
  Serial.begin(115200);
  
  if (not testing) {
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true); // enables debugging
  }
  
  delay(500);

  if (not testing) {

  if (!ELM_PORT.connect(btaddress))
  {
    Serial.println("Mdeb: \tCouldn't connect to OBD scanner - Phase 1");
      delay(READDELAY);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000)) // debug = true
  {
    Serial.println("Mdeb: \tCouldn't connect to OBD scanner - Phase 2");
    delay(READDELAY);
    Serial.println("Mdeb: \tNOT");
  }
  
  Serial.println("Mdeb: \tConnected to ELM327");
  }
}

void loop() {

if (testing) {
    // for testing display only, faking readings here //
    Serial.println(F("Mdeb: \tTesting configured"));
    soot = random(0,99);
    km = random(0,500);
    rpm = random (750,3500);
}

else {  
// read counters

  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS) { rpm = (int16_t)tempRPM; }
  Serial.print("rpm "); Serial.println(rpm);

  float tempSoot = myELM327.processPID(34, 13162, 1, 1);
  if (myELM327.nb_rx_state == ELM_SUCCESS) { soot = (int16_t)tempSoot; }
  Serial.print("% "); Serial.println(soot);

  float tempKm = myELM327.processPID(34, 12345, 1, 1);
  if (myELM327.nb_rx_state == ELM_SUCCESS) { km = (int16_t)tempKm; }
  Serial.print("km "); Serial.println(km);

}

// reading ends, now display it all / blink screen

 secondScreen();
  delay(700);
  
}

// ended here
void secondScreen (){
// displaying values      
}

```

PowerBroker2, 2024-12-06:
Since the 3.X version of the code is non-blocking, you can't call the queries for multiple PIDs one after another before ensuring you've either parsed a response or errored one PID at a time (if that makes any sense). Go back to the updated examples to see how to properly query multiple PIDs with the 3.X code.

closed 2025-02-09 by PowerBroker2