PowerBroker2/ELMduino #265: ESP32 successful connection using Mac address

MFAISALREHMAN, 2024-11-15:
Hello
ESP32 board does not connect with OBDII string name. Previously i used to connect with HC-05 and Nano but HC-05 is fragile.
Now I tried (some 100 tries) with esp32 and following code worked.

`#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

uint8_t address[6] = { 0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03 };
bool connected;

#define BT_DISCOVER_TIME 500000
esp_spp_sec_t sec_mask = ESP_SPP_SEC_NONE; // or ESP_SPP_SEC_ENCRYPT|ESP_SPP_SEC_AUTHENTICATE to request pincode confirmation
esp_spp_role_t role = ESP_SPP_ROLE_MASTER; // or ESP_SPP_ROLE_MASTER

void setup() {
  Serial.begin(38400);
  //SerialBT.setPin(pin);
  SerialBT.begin("ESP32", true, 10000);

  //SerialBT.print()
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  connected = SerialBT.connect(address, sec_mask, role);

  if (connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while (!SerialBT.connected(10000)) {
Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
    }
  }

SerialBT.print(( String) "AT Z");
SerialBT.print((const char) '\r');
delay(150);
printBT();

SerialBT.write((const uint8_t*) "AT E0",5);
delay(150);
printBT();

SerialBT.write((const uint8_t*) "AT L0",5);
delay(150);
printBT();

SerialBT.write((const uint8_t*) "AT S0",5);
delay(150);
printBT();

SerialBT.write((const uint8_t*) "AT SP 0",7);
delay(150);
printBT();

Serial.println("0100");
SerialBT.print(( String) "0100");
Serial.println("0100");

Serial.println("delay");
printBT();
delay(6000);
//SerialBT.print((const char) '\r');

printBT();
delay(1000);
}

void loop() {

SerialBT.write((const uint8_t*) "01051",5);
Serial.println("sending 01051");
delay(200);
printBT();

SerialBT.write((const uint8_t*) "0105",4);
Serial.println("sending 0105");
delay(200);
printBT();

SerialBT.write((const uint8_t*) "AT DP",5);
delay(200);
Serial.println("Protocol");
printBT();

}

void printBT()
{
  while(SerialBT.available())
 
  {
    char c = SerialBT.read();

    if(c == '>')
      Serial.println();

    Serial.write(c);
  }

}`

Consequently the ELMDuino library Initialize funtion was changed as following

`bool ELM327::initializeELM(const char &protocol, const byte &dataTimeout)
{
    char command[10] = {'\0'};
    connected = false;

   // sendCommand_Blocking(SET_ALL_TO_DEFAULTS);
    //delay(100);

    sendCommand_Blocking(RESET_ALL);
    delay(100);

    sendCommand_Blocking(ECHO_OFF);
    delay(100);

    sendCommand_Blocking(LINEFEEDS_OFF);
    delay(100);

    sendCommand_Blocking(PRINTING_SPACES_OFF);
    delay(100);

    // // Set data timeout
  //  sprintf(command, SET_TIMEOUT_TO_H_X_4MS, dataTimeout / 4);
  //  sendCommand_Blocking(command);
   // delay(100);

    // Automatic searching for protocol requires setting the protocol to AUTO and then
    // sending an OBD command to initiate the protocol search. The OBD command "0100"
    // requests a list of supported PIDs 0x00 - 0x20 and is guaranteed to work
    if ((String)protocol == "0")
    {
        // Tell the ELM327 to do an auto protocol search. If a valid protocol is found, it will be saved to memory.
        // Some ELM clones may not have memory enabled and thus will perform the search every time.
        sprintf(command, SET_PROTOCOL_TO_H_SAVE, protocol);
        if (sendCommand_Blocking(command) == ELM_SUCCESS)
        {
            if (strstr(payload, RESPONSE_OK) != NULL)
            {
                // Protocol search can take a comparatively long time. Temporarily set
                // the timeout value to 30 seconds, then restore the previous value.
                uint16_t prevTimeout = timeout_ms;
                timeout_ms = 6000;

                int8_t state = sendCommand_Blocking("0100");

                if (state == ELM_SUCCESS)
                {
                    timeout_ms = prevTimeout;
                    connected = true;
					delay(1000);
                    return connected;
                }
                else if (state == ELM_BUFFER_OVERFLOW)
                {
                    while (elm_port->available())
                        elm_port->read();
                }

                timeout_ms = prevTimeout;
            }
        }
    }
    else
    {
        // Set protocol
        sprintf(command, TRY_PROT_H_AUTO_SEARCH, protocol);

        if (sendCommand_Blocking(command) == ELM_SUCCESS)
        {
            if (strstr(payload, RESPONSE_OK) != NULL)
            {
                connected = true;
                return connected;
            }
        }
    }

    if (debugMode)
    {
        Serial.print(F("Setting protocol via "));
        Serial.print(TRY_PROT_H_AUTO_SEARCH);
        Serial.print(F(" did not work - trying via "));
        Serial.println(SET_PROTOCOL_TO_H_SAVE);
    }

    // Set protocol and save
    sprintf(command, SET_PROTOCOL_TO_H_SAVE, protocol);

    if (sendCommand_Blocking(command) == ELM_SUCCESS)
    {
        if (strstr(payload, RESPONSE_OK) != NULL)
        {
            connected = true;
            return connected;
        }
    }
    
    if (debugMode)
    {
        Serial.print(F("Setting protocol via "));
        Serial.print(SET_PROTOCOL_TO_H_SAVE);
        Serial.println(F(" did not work"));
    }

    return connected;
}`

MFAISALREHMAN, 2024-11-15:
While in ESP32 Setup function, following code may be written for connection.

` SerialBT.begin("ESP32", true, true);
  Serial.println("The device started in master mode, make sure remote BT device is on!");
 //SerialBT.setPin(pin);
  connected1 = SerialBT.connect(address, sec_mask, role);

  if (connected1) {
    Serial.println("Connected to ELM327 Bluetooth Succesfully!");

    
  } else {
    while (!SerialBT.connected(10000)) {
Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
    }
  }

  if (!myELM327.begin(ELM_PORT, DEBUG))
  {
    Serial.println("Couldn't connect to OBD scanner");
oled.println("Couldn't connect to OBD scanner");
//delay(10000);
    if (HALT_ON_FAIL)
      while (1);
  }
Serial.println("Connected to ELM327");`

MFAISALREHMAN, 2024-11-15:
link for code
[https://mega.nz/folder/2Dh0gawB#1AA_W6Qnryfu4kMNVcRypg](url)

jimwhitelaw, 2024-11-18:
I've reformatted your posts for code readability. I think you are suggesting a change to help with establishing a BT connection using MAC address, is that correct? If so, please submit your change as a pull request to facilitate the evaluation of the change. thx

dani9084, 2024-12-22:
@MFAISALREHMAN 
Could you please provide me with the version of ELMduino and esp32 board manager that you are using?

MFAISALREHMAN, 2024-12-25:
I have made lil changes to suit my need but it did connect to esp32 very fast.  you have to copy the folder content below i am sharing to this location ( C:\Users\DELL153000\Documents\Arduino\libraries) of pc. 

https://mega.nz/folder/KOoGib4b#3WvxzgEzgrv9pPKV9oQiEg

MFAISALREHMAN, 2024-12-25:
I am not familiar with pull request,  but i change lil bit the ELM327::initializeELM in order to connect to ESP32 (may be Chinese version),  e.g the timeout of 30000 is more for searching protocol. Mine esp32 disconnects in 9 second.  In 9s, esp timeout, if a request for PID is not sent.

dani9084, 2024-12-25:
Do you still use v2.0.17 of the esp32 board manager or have you updated to v3.x? Thanks

MFAISALREHMAN, 2024-12-26:
I did not update any firmware of esp32, i am using the stock version that was delivered to me. You can tell me where to check the boot manager.

MFAISALREHMAN, 2025-03-18:
https://mega.nz/file/HW4xWRCa#OlVzH3OkX-eecowBCegpLbKhnE8op4LFUWUbjRULrgs

you can download the code above but customize it for your need.
For this you have to change lil bit of intitialization code code in ELMduino Library files (.cpp) . See the second part / code of initial post.

MFAISALREHMAN, 2025-03-18:
The library files are location are C:\Users\{username}\Documents\Arduino\Elmduino

lgarcia33166, 2025-03-18:
Thank you so much for your help, I will try it.

Kind regards,
Luis García