PowerBroker2/ELMduino #268: What's the average time between queries?

doublep21, 2024-12-22:
i'm building a 3 display hud for my car and in the first test i saw a low response time doing 6 queries sequentially , so whats the avg response time i can expect and there is any way to improve it?

PowerBroker2, 2024-12-23:
I don't have any statistics on query response speeds, but the library is setup to minimize response time already. I would say ensure your sketch is as efficient and responsive as possible. Reducing the number of PIDs queried should also help if you can get away with it. The biggest delays should be due to the ELM327 firmware itself - something I have no real insight into

doublep21, 2024-12-23:
> I don't have any statistics on query response speeds, but the library is setup to minimize response time already. I would say ensure your sketch is as efficient and responsive as possible. Reducing [...]

Thank you ☺️

closed 2024-12-23 by doublep21

jimwhitelaw, 2024-12-23:
I know this is closed now, but I’m adding some info that may be useful for someone. The OBD2 standard query rate limit is 20 / sec. That doesn’t mean that every adapter or ecu will support that rate. In my projects, I typically limit my queries to 10/sec to be safe. 

As to smooth updating of your displays, I can offer a few tips. One option is to use separate cpu cores for updating the data and for updating your displays. There’s an example of this provided in the examples folder. See also the multiple PID examples for the correct way to implement your main loop and if you do need some kind of timer, create your own timeout check and DO NOT use delay(). In my projects, I typically create a shared data struct that is written to by a sensor method and read from by a display method. This can mask temporary slow updates as the display code is never waiting on a response and it also makes it easier if you want to display moving average values to smooth out fluctuations, etc.

doublep21, 2024-12-24:
> I know this is closed now, but I’m adding some info that may be useful for someone. The OBD2 standard query rate limit is 20 / sec. That doesn’t mean that every adapter or ecu will support that [...]

[image]
that is my work so far , in my code there is no delay whatsoever but for now i'm using only one core and maybe it's enough . I will query more often important and instantaneous stuff like speed , rpm and engine load and the rest after few seconds

ligius-, 2025-05-14:
I don't want to open a new issue or question since I'm in the same boat. Trying to get some EV data which works fine in non-blocking mode when doing things very slowly. However, once I start to speed up the queries (10/s) I get timeouts in the blocking function. I still get the correct response most of the times but other times response data from previous queries is mixed with response from the last query.

Ideally I would have these simultaneous measurements as the multiplication needs to be performed on-time: battery power, battery voltage, rear motor torque, rear motor rpm, front motor torque, front motor rpm. This doesn't seem to be doable to get data at a real-time pace.

Let's say I split the timings for the queries: battery current and voltage every 0.1s, motors every 1s, rest of data every 10s. What would be the best way to achieve this and not run into the issues that I have been running into? I assume I need to launch a query, wait (yield) until the response and do this for every data point. Don't even know if it's possible to query multiple or have a stream.

PowerBroker2, 2025-05-14:
You can only do one query at a time - this is a limitation of the OBD2/ELM interface. I would say experimentation on your vehicle would be the best way to figure out what timing is right for you and your project.

ligius-, 2025-05-17:
Using a simple test:

```
  myELM327.sendCommand_Blocking("22480d");  // Send the PID commnad

  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    double hvVoltage = myELM327.conditionResponse(calcHVVoltage);
    Serial.print(hvVoltage);
    Serial.print("V ");

    long elapsed = millis() - lastReadTime;
    lastReadTime = millis();

    Serial.printf(" rate %d/s \n", 1000 / elapsed);
  } else {
    myELM327.printError();
    delay(2000);
  }
```

I get about 14-15 queries per second with no error. If I query two PIDs I get exactly 7/s. Just wanted to share this in case someone is wondering how fast it can go. I did not receive any errors or timeouts, so my previous timeout behavior is probably linked to LVGL interference.

doublep21, 2025-05-17:
> Using a simple test:
>
> ```
>   myELM327.sendCommand_Blocking("22480d");  // Send the PID commnad
>
>   if (myELM327.nb_rx_state == ELM_SUCCESS) {
>     double hvVoltage = [...]

you're lucky ,  i have an older car (2003) and it's quite slower , maybe 4 querys at second approximately

OsIrIs29000, 2025-06-23:
> [image] that is my work so far , in my code there is no delay whatsoever but for now i'm using only one core and maybe it's enough . I will query more often important and instantaneous stuff like [...]

Hi, I also have slowness issues and so I was looking for answers to my questions on this topic, and I found your gauges very beautiful, did you find them on another project or did you make them yourself?

OsIrIs29000, 2025-06-23:
> > > > I know this is closed now, but I’m adding some info that may be useful for someone. The OBD2 standard query rate limit is 20 / sec. That doesn’t mean that every adapter or ecu will support that rate. In my projects, I typically limit my queries to 10/sec to be safe.
> > > > As to smooth updating of your displays, I can offer a few tips. One option is to use separate cpu cores for updating the data and for updating your displays. There’s an example of this provided in the examples folder. See also the multiple PID examples for the correct way to implement your main loop and if you do need some kind of timer, create your own timeout check and DO NOT use delay(). In my projects, I typically create a shared data struct that is written to by a sensor method and read from by a display method. This can mask temporary slow updates as the display code is never waiting on a response and it also makes it easier if you want to display moving average values to smooth out fluctuations, etc.
> > > 
> > > 
> > > [image] that is my work so far , in my code there is no delay whatsoever but for now i'm using only one core and maybe it's enough . I will query more often important and instantaneous stuff like speed , rpm and engine load and the rest after few seconds
> > 
> > 
> > Hi, I also have slowness issues and so I was looking for answers to my questions on this topic, and I found your gauges very beautiful, did you find them on another project or did you make them yourself?
> 
> thank you , i made it myself .
> 
> [image]

Congrats ! Can I ask you what did you use to make that ?

doublep21, 2025-06-23:
> Congrats ! Can I ask you what did you use to make that ?

yeah sure , i used arduino espi , lvgl and square line studio

ligius-, 2025-06-24:
Offtopic, but I used EEZ Studio instead, I don't like the shenanigans that Squareline pulled with the license and their limitations - which they can also change at any time.

doublep21, 2025-07-03:
@OsIrIs29000 

[image]

OsIrIs29000, 2025-07-03:
> > > [@OsIrIs29000](https://github.com/OsIrIs29000)
> > > [image]
> > 
> > 
> > Wonderful, still havent manage to make one, my gc9a01 lcd display doesn't seem to be able to display things using lvlg library, and I havent found any solutions for the high delays between my elm327 and esp32 😮‍💨
> > can you share some code?

Do you have a Discord ?

OsIrIs29000, 2025-07-04:
> > ?
> 
> yea , give me your username

It should be 0slrls