PowerBroker2/ELMduino #270: ELMduino BLE Support

vdvornichenko, 2024-12-29:
Hi there,

I built a small Serial wrapper for BLE connection, so now it's possible to use ELMduino with BLE 4.0 OBD dongles.

Tested with OBDLink CX, custom headers and PID are also working fine, you can take a look.

https://github.com/vdvornichenko/obd-ble-serial

patfelst, 2024-12-30:
Amazing, great work - well done!

jimwhitelaw, 2025-01-12:
This is excellent! I started on something similar but it got sidetracked for various reasons and never finished. Your implementation looks more robust than what I was working on. Nice. This could be useful for other projects as well that need a BLE to Serial library. I'm looking forward to trying this out soon.

Ldalvik, 2025-01-21:
Hello, what devices have you tried this on? I tried multiple ESP32S3 Dev Module LCD screens and they all crash when using your example.

ElettroshockNow, 2025-03-13:
you are a legend .... I was tired of always having to enter the MAC .... thank you thank you and a thousand thanks

Saluti
Luca

ElettroshockNow, 2025-03-14:
For the corrupted data passed to elmduino (which still manages to interpret well), I took the liberty of modifying the library

``````
static void notifyCallback(
    BLERemoteCharacteristic *pBLERemoteCharacteristic,
    uint8_t *pData,
    size_t length,
    bool isNotify)
{   
    //Serial.print("[DEBUG] ELM RESPONSE > ");
    //printFriendlyResponse(pData, length);

    // Convertiamo i dati ricevuti in una stringa
    std::string receivedData((char*)pData, length);

    // Se la nuova stringa è già presente alla fine di staticBuffer, evitiamo di concatenarla
    if (staticBuffer.size() < length || staticBuffer.substr(staticBuffer.size() - length) != receivedData) {
        staticBuffer += receivedData;
    }

    //Serial.print("[DEBUG] staticBuffer after append: ");
    //Serial.println(staticBuffer.c_str());
}
``````
printout without [DEBUG] ELM RESPONSE   (ESP32 ELMduino 3.3.2 OBD Vlink connected via BLE )
``````
22:23:00.793 Sending the following command/query: AT RV
22:23:00.846 	Received char: A
22:23:00.866 	Received char: T
22:23:00.886 	Received char: _
22:23:00.906 	Received char: R
22:23:00.927 	Received char: V
22:23:00.946 	Received char: \r
22:23:00.966 	Received char: 1
22:23:00.986 	Received char: 3
22:23:01.006 	Received char: .
22:23:01.026 	Received char: 6
22:23:01.046 	Received char: V
22:23:01.066 	Received char: \r
22:23:01.087 	Received char: \r
22:23:01.110 	Received char: >
22:23:01.110 Delimiter found.
22:23:01.110 All chars received: ATRV13.6V
22:23:01.112 Battery Voltage: 13.60
22:23:01.130 Service: 1
22:23:01.130 PID: 12
22:23:01.130 Normal length query detected
22:23:01.130 Query string: 010C1
22:23:01.135 Clearing input serial buffer
22:23:01.135 Sending the following command/query: 010C1
22:23:01.205 	Received char: 0
22:23:01.225 	Received char: 1
22:23:01.245 	Received char: 0
22:23:01.265 	Received char: C
22:23:01.285 	Received char: 1
22:23:01.305 	Received char: \r
22:23:01.325 	Received char: 4
22:23:01.345 	Received char: 1
22:23:01.365 	Received char: 0
22:23:01.385 	Received char: C
22:23:01.405 	Received char: 0
22:23:01.425 	Received char: C
22:23:01.445 	Received char: D
22:23:01.465 	Received char: 8
22:23:01.485 	Received char: \r
22:23:01.505 	Received char: \r
22:23:01.529 	Received char: >
22:23:01.529 Delimiter found.
22:23:01.529 All chars received: 010C1410C0CD8
22:23:01.534 Expected response header: 410C
22:23:01.534 Single response detected
22:23:01.539 	Processing hex nibble: 0
22:23:01.539 	Processing hex nibble: C
22:23:01.539 	Processing hex nibble: D
22:23:01.545 	Processing hex nibble: 8
22:23:01.545 64-bit response: 
22:23:01.551 	responseByte_0: 216
22:23:01.551 	responseByte_1: 12
22:23:01.551 	responseByte_2: 0
22:23:01.557 	responseByte_3: 0
22:23:01.557 	responseByte_4: 0
22:23:01.557 	responseByte_5: 0
22:23:01.557 	responseByte_6: 0
22:23:01.560 	responseByte_7: 0
22:23:01.560 rpm: 822
``````

for the restart problem when it doesn't find odb....I don't know where to start :)

ElettroshockNow, 2025-03-19:
**SOLVED** "LoadProhibited" panic.....i added a mydevice check
``````
bool BLEClientSerial::connect(void)
{
    Serial.println("Forming a connection to ");
    // Controllo che myDevice sia stato trovato durante la scansione BLE
    if (myDevice == nullptr) {
        Serial.println("Errore: Nessun dispositivo BLE trovato! Assicurati che l'OBD sia acceso.");
        return false;
    }
    Serial.println(myDevice->getAddress().toString().c_str());
``````

Erictric, 2025-10-31:
Great work! I was finally able to update my WT32-SC01 board to the WT32-SC01-Plus which doesn't handle bluetooth classic. I'm using it for a OBD2 display for my Hyundai Ioniq5. Big thank you!