PowerBroker2/ELMduino #272: State of Charge - BYD

dgaust, 2025-01-27:
Hi,

I'm trying to parse the state of charge from a BYD Seal using an ODBLink CX+. I know ELMduino doesn't support BLE out of the box, but using a BLE solution someone pasted here, I'm able to communicate with the dongle and retrieve data. I'm just having trouble with parsing the response from the SOC.

The SoC can be queried using the PID 221FFC and I have to send ATSH7E7 to get the right header (or ECU, whatever the right terminology is).

The response comes back like

```
[DEBUG] ELM RESPONSE > \r>
	Received char: 7
	Received char: E
	Received char: F
	Received char: 0
	Received char: 7
	Received char: 6
	Received char: 2
	Received char: 1
	Received char: F
	Received char: F
	Received char: C
	Received char: 6
	Received char: 2
	Received char: 1
	Received char: 6
	Received char: 6
	Received char: 0
	Received char: 3
	Received char: B
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 7EF07621FFC6216603B
```

I've figured out if that what I'm looking for is after the after the 7EF07621FFC, as the 62 and 16 - as when these are converted from HEX and put into the formula (A +B*256)/100, it equals the current SoC from the car.

So hoping someone can help with some tips on how to parse that from the response that ELMduino gets from the device. I've hacked together the below from the examples and discussions here, but I'm at a little bit of a loss so any advice appreciated.

```
#include <M5Unified.h>
#include <M5GFX.h>
#include <M5Dial.h>
#include <Arduino.h>
#include "ELMduino.h"
#include <BLEClientSerial.h>

BLEClientSerial BLESerial;

#define DEBUG_PORT Serial
#define ELM_PORT   BLESerial

ELM327 myELM327;

typedef enum { BATT_VOLTAGE,
               STATE_OF_CHARGE,
               BATT_TEMP } obd_pid_states;

obd_pid_states obd_state = STATE_OF_CHARGE;
int nb_query_state = SEND_COMMAND;

void setup()
{
  M5Dial.begin();
  //Set font color
  M5Dial.Display.setTextColor(GREEN);
  //Set font alignment
  M5Dial.Display.setTextDatum(middle_center);
  //Set font
  M5Dial.Display.setTextFont(&fonts::Orbitron_Light_32);
  //Set font size
  M5Dial.Display.setTextSize(1);
  //Print text
  M5Dial.Display.drawString("Connecting", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("OBDLink CX");

  if (!ELM_PORT.connect())
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
  M5Dial.Display.clearDisplay(TFT_BLACK);
  M5Dial.Display.drawString("Connected", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);
  myELM327.sendCommand_Blocking("ATZ");
  myELM327.sendCommand_Blocking("ATD");
  myELM327.sendCommand_Blocking("ATD0");
  myELM327.sendCommand_Blocking("ATH1");
  myELM327.sendCommand_Blocking("ATSP6");
  myELM327.sendCommand_Blocking("ATE0");
  myELM327.sendCommand_Blocking("ATH1");
  myELM327.sendCommand_Blocking("ATM0");
  myELM327.sendCommand_Blocking("ATS0");
  myELM327.sendCommand_Blocking("ATAT1");
  myELM327.sendCommand_Blocking("ATAL");
  myELM327.sendCommand_Blocking("STCSEGT1");
  myELM327.sendCommand_Blocking("ATST96");
  myELM327.sendCommand_Blocking("ATSH7E7");
}

void loop()
{
  switch (obd_state)
  {
    case BATT_VOLTAGE:
    {
      float battVoltage  = myELM327.batteryVoltage();
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        Serial.print("Battery Voltage: "); Serial.println(battVoltage);
        // M5Dial.Display.clearDisplay(TFT_BLACK);
        M5Dial.Display.drawString(String(battVoltage), M5Dial.Display.width() / 2, M5Dial.Display.height() - 50 );
        obd_state = STATE_OF_CHARGE;
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
        obd_state = STATE_OF_CHARGE;
      }
      break;
    }

    case STATE_OF_CHARGE:
    {
      if (nb_query_state == SEND_COMMAND)      
      {
        myELM327.sendCommand("221FFC");         
        nb_query_state = WAITING_RESP;     
      }
      else if (nb_query_state == WAITING_RESP)    
      {
        myELM327.get_response();                
      }
    
      if (myELM327.nb_rx_state == ELM_SUCCESS)    
      { 
        
        int A = myELM327.payload[8];
        int B = myELM327.payload[9];
        B = B * 256;           
        float temperature = (A + B) / 100;          
        Serial.print("soc a: "); Serial.println(temperature);
        M5Dial.Display.drawString(String(temperature), M5Dial.Display.width() / 2, 50);
        nb_query_state = SEND_COMMAND;
        delay(5000);      
        obd_state = BATT_TEMP;                     
      }
    
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {                                           
        nb_query_state = SEND_COMMAND;         
        myELM327.printError();
        delay(5000);                           
      }
      break;
    }

    case BATT_TEMP:
    {
      
      if (nb_query_state == SEND_COMMAND)       
      {
        myELM327.sendCommand("220032");        
        nb_query_state = WAITING_RESP; 
      }
      else if (nb_query_state == WAITING_RESP)
      {
        myELM327.get_response();
      }
    
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      { 
        int A = myELM327.payload[5]; 
        float temperature = A - 40;
        Serial.print("Battery temp: "); Serial.println(temperature);
        M5Dial.Display.drawString(String(temperature), M5Dial.Display.width() / 2, 85);
        nb_query_state = SEND_COMMAND;
        delay(5000);      
        obd_state = BATT_VOLTAGE;
      }
    
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {                                           
        nb_query_state = SEND_COMMAND;          
        myELM327.printError();
        delay(5000);                           
      }
      break;
    }
  }
}
```

jimwhitelaw, 2025-01-27:
It looks like you are most of the way there. The missing piece is that two hex values from the payload need to be combined into a single int value for both A & B. Also it appears that the code is using the wrong bytes from _payload[]_. 

Try this (untested!) snippet instead: 

```C
        int A = (ctoi(myELM327.payload[10]) << 4) | ctoi(myELM327.payload[11])];
        int B = (ctoi(myELM327.payload[12]) << 4) | ctoi(myELM327.payload[13]);  
        B = B * 256;
        float soc = (A + B) / 100;          
        Serial.print("soc a: "); Serial.println(soc);
        M5Dial.Display.drawString(String(soc), M5Dial.Display.width() / 2, 50);
        nb_query_state = SEND_COMMAND;
        delay(5000);      
        obd_state = BATT_TEMP;
```

Also, that's good news that BLE is working for you. I think that compatibility will become increasingly important as many new devices and dev boards only support BLE now.

dgaust, 2025-01-27:
Thanks, that works with some minor tweaks, just to offset the payload by 1, and remove a typo. I grabbed the ctoi bit from another custom PID post, and it seems to work as expected

```
int A = (ctoi(myELM327.payload[11]) << 4) | ctoi(myELM327.payload[12]);
int B = (ctoi(myELM327.payload[13]) << 4) | ctoi(myELM327.payload[14]); 
```

Now to have it displayed with the decimal, and not just a rounded int :D

closed 2025-02-09 by PowerBroker2