PowerBroker2/ELMduino #273: A small problem

toprakb66, 2025-02-09:
Hello my friend, I have been trying to read data from the OBD for a long time, around 3-4 months, and I have tried many projects, but the results were negative. I have made the best progress with your project. It's currently 3 AM and I'm still working on it. I will ask my question now, but if you're still active, I will go to the car and take a screenshot of the error on the serial monitor.

toprakb66, 2025-02-10:
Bro, my code is the same, and I’ve tested it with both the Wi-Fi and Bluetooth modules, but I’m getting the same results on the serial monitor with both. What could be the problem? Please help me. `#include <WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi_OBDII";
const char* password = "your-password";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);
WiFiClient client;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_AP);
  WiFi.begin(ssid);
  // WiFi.begin(ssid, password); //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000))
    Serial.println("connected");
  else
  {
    Serial.println("connection failed");
    while(1);
  }

  myELM327.begin(client, true, 2000);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}`

toprakb66, 2025-02-10:
SERİAL SCREEN `Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: A
	Received char: T
	Received char: _
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char: _
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
Received char: A
	Received char: T
	Received char: _
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT SP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: 0100
	Received char: S
	Received char: E
	Received char: A
	Received char: R
	Received char: C
	Received char: H
	Received char: I
	Received char: N
	Received char: G
	Received char: .
	Received char: .
	Received char: .
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 8
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: SEARCHING...410080000000
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 7
	Received char: F
	Received char: 0
	Received char: 1
	Received char: 1
	Received char: 2
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 7F0112
Expected response header: 410C
Response not detected
WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
RPM: 0
Service: 1
PID: 12
Normal length query detected`

toprakb66, 2025-02-10:
serial monitor  

Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: A
	Received char: T
	Received char: _
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char: _
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
Received char: A
	Received char: T
	Received char: _
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT SP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: 0100
	Received char: S
	Received char: E
	Received char: A
	Received char: R
	Received char: C
	Received char: H
	Received char: I
	Received char: N
	Received char: G
	Received char: .
	Received char: .
	Received char: .
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 8
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: SEARCHING...410080000000
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 7
	Received char: F
	Received char: 0
	Received char: 1
	Received char: 1
	Received char: 2
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 7F0112
Expected response header: 410C
Response not detected
WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
RPM: 0
Service: 1
PID: 12
Normal length query detected

PowerBroker2, 2025-02-10:
Bro, please use proper markdown formatting for your code and debug prints, please.

From what I can tell, however, is that your car/ELM327 isn't responding properly to the queries and is not a library issue. What is your car's make and model?

toprakb66, 2025-02-10:
Hello my friend, first of all, thank you for creating such a great project and taking the time to respond to me. I have reviewed the closed question articles, but I saw that you mentioned it doesn’t work on vehicles older than 2008. My car is a 2006 Kia Sorento, and it uses the ISO 14230-4 KWP protocol. I was able to connect via a module using my phone. Is there any other way to make this system work on my car?

Please guide me on how I can get this system to work in my vehicle. This project is very important to me.

toprakb66, 2025-02-10:
[image] My friend, you definitely need to see this! There's no mistake here, the response is normal because the vehicle is in stop mode. This excites me, I took the photo this morning and just noticed it now

toprakb66, 2025-02-10:
Dude, I can't believe my eyes! After struggling for so long, my project has finally shown some light, and I successfully received this response with the WiFi module. My next goal is to get the temperature data. It's late now, so I'll test it tomorrow. I'm so happy my hands are shaking! :) `Normal length query detected
"Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: E
	Received char: 5
	Received char: C
	Received char: \r
	Received char: \r
	Received char: >
Delimiter `found."

toprakb66, 2025-02-10:
Dude, I've been working on this for 4 months. Even though not every day, I kept going back and forth to the car, getting negative results. This feels like a miracle to me! Thank you so much for the effort you've put into this project. I tested it, and the code is the one that reads multiple PIDs.

toprakb66, 2025-02-11:
> #include <WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi_OBDII";  // ELM327 Wi-Fi SSID'si
const char* password = "";        // Eğer şifre varsa buraya ekle

IPAddress server(192, 168, 0, 10);  // ELM327'nin IP adresi
WiFiClient client;
ELM327 myELM327;

typedef enum { ENG_RPM, COOLANT_TEMP } obd_pid_states;
obd_pid_states obd_state = ENG_RPM;

float rpm = 0;
float coolantTemp = 0;

void setup()
{
    Serial.begin(115200);
    
    Serial.print("Wi-Fi'ye bağlanılıyor: ");
    Serial.println(ssid);

    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid);  // Şifre varsa kullan

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    Serial.println("\nWi-Fi bağlantısı tamamlandı.");
    Serial.print("Cihaz IP Adresi: ");
    Serial.println(WiFi.localIP());

    // ELM327'ye bağlan
    if (!client.connect(server, 35000))
    {
        Serial.println("ELM327 bağlantı hatası!");
        while (1);
    }
    else
    {
        Serial.println("ELM327 bağlantısı başarılı.");
    }

    if (!myELM327.begin(client, true, 2000))
    {
        Serial.println("ELM327 başlatılamadı!");
        while (1);
    }

    Serial.println("ELM327 hazır.");
}

void loop()
{
  switch (obd_state)
  {
    case ENG_RPM:
    {
      rpm = myELM327.rpm();
      
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        Serial.print("RPM: ");
        Serial.println(rpm);
        obd_state = COOLANT_TEMP;
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
        obd_state = COOLANT_TEMP;
      }
      
      break;
    }
    
    case COOLANT_TEMP:
    {
      int rawTemp = myELM327.engineCoolantTemp();  // OBD-II'den su sıcaklığını oku
      
      if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
        coolantTemp = rawTemp - 40; // OBD-II formülü: A - 40
        Serial.print("Soğutma Suyu Sıcaklığı: ");
        Serial.print(coolantTemp);
        Serial.println(" °C");
        obd_state = ENG_RPM;
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
        obd_state = ENG_RPM;
      }
      
      break;
    }
  }
}

>  Can you take a look at this code? I modified it to get the temperature data, but since my car isn't here, I can't test it. Could you check if there's any issue or not

jimwhitelaw, 2025-02-28:
I didn't test it myself, but codellama says "the code is correct and should work as expected", fwiw.