PowerBroker2/ELMduino #276: Multiple PIDs - can't get working

Dropsik, 2025-02-20:
Hardware: Generic devboard ESP32-Wroom-32 with oled
OBD Adapter: Vgate iCar Pro 2S

Code:
`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128  // OLED display width, in pixels
#define SCREEN_HEIGHT 64  // OLED display height, in pixels
#define OLED_RESET -1     // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial
String Obd_name = "Android-Vlink"; //Android-Vlink or OBDII
const char *Obd_pin = "1234"; 

ELM327 myELM327;

uint32_t rpm = 0;
enum QueryData {
  UPDATE_SCREEN, GET_DPF_PERCENT, READ_DPF_PERCENT, GET_DPF_SMC, READ_DPF_SMC, GET_DPF_KM, READ_DPF_KM
  };

QueryData queryData = GET_DPF_PERCENT;

float dpfPercent = -1;
float dpfSmc = -1;
float dpfKm = -1;

uint8_t ctoi(uint8_t value) {
    if (value >= 'A')
        return value - 'A' + 10;
    else
        return value - '0';
}

void setup() {

  if (!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) {
    Serial.println(F("SSD1306 allocation failed"));
    for (;;)
      ;
  }
  display.clearDisplay();
  display.setTextSize(2);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  display.println("DPF Reader");
  display.setTextSize(1);
  display.println("Initializing ...");
  display.println("Name: " + Obd_name);
  display.println("PIN: " + String(Obd_pin));
  display.display();
  display.setTextSize(1);
  display.setCursor(0, 0);

#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  SerialBT.setPin(Obd_pin);
  ELM_PORT.begin("DPFreader", true);

  if (!ELM_PORT.connect(Obd_name)) {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    display.clearDisplay();
    display.println("Couldn't connect to OBD scanner - Phase 1");
    display.display();
    while (1)
      ;
  }

  if (!myELM327.begin(ELM_PORT, true, 2000)) {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    display.clearDisplay();
    display.println("Couldn't connect to OBD scanner - Phase 2");
    display.display();
    while (1)
      ;
  }

  Serial.println("Connected to ELM327");
  display.clearDisplay();
  display.println("Connected to ELM327");
  display.display();
  delay(1000);
  // set header
  myELM327.sendCommand_Blocking("ATSH 7E0");
}

void loop() {
  
  switch (queryData) {
    case GET_DPF_PERCENT:    
      myELM327.sendCommand("22115B");
      Serial.println("Get DPF Percent");
      queryData = READ_DPF_PERCENT;
      break;
    case READ_DPF_PERCENT:
      myELM327.get_response();
      if (myELM327.nb_rx_state == ELM_SUCCESS) {
          Serial.println("DPF percent response received.");
          int A = (ctoi(myELM327.payload[6]) << 4) + ctoi(myELM327.payload[7]);
          dpfPercent = A;
          Serial.println(dpfPercent);
          queryData = GET_DPF_SMC;
      } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
          Serial.println("Error receiving DPF Percent data.");
          myELM327.printError();
          queryData = READ_DPF_PERCENT;
      }
      break;
    case GET_DPF_SMC:
      Serial.println("Get DPF SMC");
      myELM327.sendCommand("22114F");
      queryData = READ_DPF_SMC;
      break;
    case READ_DPF_SMC:
      if (myELM327.nb_rx_state == ELM_SUCCESS) {
          Serial.println("DPF SMC:");
          int A = (ctoi(myELM327.payload[6]) << 4) | ctoi(myELM327.payload[7]);
          int B = (ctoi(myELM327.payload[8]) << 4) | ctoi(myELM327.payload[9]);
          dpfSmc = ((A*256) + B) / 100;
          Serial.println(dpfSmc);
          queryData = GET_DPF_KM;
      } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
          Serial.println("Error receiving DPF SMC.");
          myELM327.printError();
          queryData = READ_DPF_SMC;
      }
      break;
    case GET_DPF_KM:
      Serial.println("Get DPF KM");
      myELM327.sendCommand("221156");
      queryData = READ_DPF_KM;
      break;
    case READ_DPF_KM:
      if (myELM327.nb_rx_state == ELM_SUCCESS) {
          Serial.println("DPF KM:");
          int A = (ctoi(myELM327.payload[6]) << 4) | ctoi(myELM327.payload[7]);
          int B = (ctoi(myELM327.payload[8]) << 4) | ctoi(myELM327.payload[9]);
          int C = (ctoi(myELM327.payload[10]) << 4) | ctoi(myELM327.payload[11]);
          int D = (ctoi(myELM327.payload[12]) << 4) | ctoi(myELM327.payload[13]);
          dpfKm = (A + B + C + D) / 1000;
          Serial.println(dpfKm);
          queryData = UPDATE_SCREEN;
      } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
          Serial.println("Error receiving DPF SMC.");
          myELM327.printError();
          queryData = READ_DPF_KM;
      }
      break;
    case UPDATE_SCREEN:
      updateScreen();
      queryData = GET_DPF_PERCENT;
      break;
  }
delay(500);
Serial.println("MainLoop.");

}

void updateScreen() {
  display.clearDisplay();
  display.setTextSize(2);
  display.setTextColor(WHITE);
  display.setCursor(0, 0);
  display.println("OFF"); // Active, Passive
  display.setTextSize(1);
  display.println("Percent: " + String(dpfPercent/100) + "%");
  display.println("Soot Mass Calc: " + String(dpfSmc));
  display.println("Distance: " + String(dpfKm));
  display.display();
  display.setTextSize(1);
  display.setCursor(0, 0);
}`

Question:
1) case GET_DPF_PERCENT and READ_DPF_PERCENT work as expected, but after GET_DPF_SMC code remain in main loop

`11:53:41.149 -> 	Received char: >
11:53:41.149 -> Delimiter found.
11:53:41.149 -> All chars received: SEARCHING...4100983BA013410098180001
11:53:41.149 -> Connected to ELM327
11:53:42.199 -> Clearing input serial buffer
11:53:42.199 -> Sending the following command/query: ATSH 7E0
11:53:42.232 -> 	Received char: O
11:53:42.232 -> 	Received char: K
11:53:42.232 -> 	Received char: \r
11:53:42.232 -> 	Received char: \r
11:53:42.232 -> 	Received char: >
11:53:42.232 -> Delimiter found.
11:53:42.232 -> All chars received: OK
11:53:42.232 -> Clearing input serial buffer
11:53:42.232 -> Sending the following command/query: 22115B
11:53:42.232 -> Get DPF Percent
11:53:42.723 -> MainLoop.
11:53:42.723 -> 	Received char: 6
11:53:43.216 -> MainLoop.
11:53:43.216 -> 	Received char: 2
11:53:43.741 -> MainLoop.
11:53:43.741 -> 	Received char: 1
11:53:44.237 -> MainLoop.
11:53:44.237 -> 	Received char: 1
11:53:44.730 -> MainLoop.
11:53:44.730 -> 	Received char: 5
11:53:45.224 -> MainLoop.
11:53:45.224 -> 	Received char: B
11:53:45.723 -> MainLoop.
11:53:45.723 -> 	Received char: 0
11:53:46.217 -> MainLoop.
11:53:46.217 -> 	Received char: 0
11:53:46.743 -> MainLoop.
11:53:46.743 -> 	Received char: \r
11:53:47.236 -> MainLoop.
11:53:47.236 -> 	Received char: \r
11:53:47.728 -> MainLoop.
11:53:47.728 -> 	Received char: >
11:53:47.728 -> Delimiter found.
11:53:47.728 -> All chars received: 62115B00
11:53:47.728 -> DPF percent response received.
11:53:47.728 -> 0.00
11:53:48.219 -> MainLoop.
11:53:48.219 -> Get DPF SMC
11:53:48.219 -> Clearing input serial buffer
11:53:48.219 -> Sending the following command/query: 22114F
11:53:48.714 -> MainLoop.
11:53:49.237 -> MainLoop.
11:53:49.731 -> MainLoop.
11:53:50.224 -> MainLoop.
11:53:50.715 -> MainLoop.
11:53:51.238 -> MainLoop.
11:53:51.729 -> MainLoop.
11:53:52.221 -> MainLoop.
11:53:52.714 -> MainLoop.
11:53:53.237 -> MainLoop.
11:53:53.730 -> MainLoop.
11:53:54.225 -> MainLoop.
11:53:54.718 -> MainLoop.
11:53:55.247 -> MainLoop.`

2) DPF percentage from CarScanner app is 42% - but above it looks like 0% (formula is straight A), so my next question is
3) How the payload is parsed for formula values: A,B,C and so on

jimwhitelaw, 2025-02-21:
1. Your READ_DPF_SMC and READ_DPF_KM blocks are missing the _myELM327.get_response()_ method calls.
2. How the data is returned in the response depends on the PID definition from the mfg. Your parsing appears to be correct, but you can see that the value returned in the response is 00. Are you certain that 7E0 is the correct ECU ID for this vehicle and PID?

Dropsik, 2025-02-21:
1. Thank you! I looked on this code half day and I don't saw that :| - I'l check it later
2. My fault, PID and answer is correct, it's not percent of load but DPF Regeneration state and returned values: 0 normal, 1 passive, 2 active - so I have to wait to check it is parsed correctly.

By the way, do you know any clever way to obtain PIDs (probably from app on phone like CarScanner) ?

jimwhitelaw, 2025-02-22:
You need to cast the operands to before dividing. 

```C
dpfSmc = (float)((A*256) + B) / 100;
```

I don t know of any easy way to find custom PID. Other than searching online for sources, like forums for a particular vehicle or maybe the Torque forums?

As for parsing the response, you need to know how many bytes are expected and work it out from there. It looks like what you are doing is correct, probably the actual float value is getting lost when dividing int values as above.

Dropsik, 2025-02-22:
Thank you, float works. But still I can't get kilometers parsed correctly. Maybe you can help with this:
```
19:05:27.204 -> Get DPF KM
19:05:27.204 -> Clearing input serial buffer
19:05:27.204 -> Sending the following command/query: 221156
19:05:27.235 -> 	Received char: 6
19:05:27.235 -> 	Received char: 2
19:05:27.235 -> 	Received char: 1
19:05:27.235 -> 	Received char: 1
19:05:27.235 -> 	Received char: 5
19:05:27.235 -> 	Received char: 6
19:05:27.235 -> 	Received char: 0
19:05:27.235 -> 	Received char: 0
19:05:27.235 -> 	Received char: 0
19:05:27.235 -> 	Received char: 1
19:05:27.235 -> 	Received char: 5
19:05:27.235 -> 	Received char: 3
19:05:27.235 -> 	Received char: D
19:05:27.235 -> 	Received char: B
19:05:27.269 -> 	Received char: \r
19:05:27.269 -> 	Received char: \r
19:05:27.269 -> 	Received char: >
19:05:27.301 -> Delimiter found.
19:05:27.301 -> All chars received: 621156000153DB
19:05:27.301 -> DPF KM:
19:05:27.301 -> 0.00
```
formula is: dpfKm = (A + B + C + D) / 1000;

Dropsik, 2025-02-22:
In this case, it can be integer. but I found something else.

[image]
When I convert whole hex it makes sense 87003 / 1000 = 87KM

jimwhitelaw, 2025-02-24:
Either of these (untested!) changes should work. The first is kinda quick and dirty keeping the existing parsing, the second probably marginally more efficient. 

```C
dpfKm = (float)(A<<24 | B<<16 | C<<8 | D) /1000;
```
```C
int A = (ctoi(myELM327.payload[6]) << 28) | 
        (ctoi(myELM327.payload[7] << 24) | 
        (ctoi(myELM327.payload[8]) << 20) | 
        (ctoi(myELM327.payload[9]) << 16) | 
        (ctoi(myELM327.payload[10]) << 12) | 
        (ctoi(myELM327.payload[11] << 8) | 
        (ctoi(myELM327.payload[12]) << 4) | 
        (ctoi(myELM327.payload[13]);

dpfKm  = (float)A/1000;
```

Dropsik, 2025-02-26:
Brilliant way and it's works like a charm!
For now I think I can close this issue. Thank you for help!

closed 2025-02-26 by Dropsik