PowerBroker2/ELMduino #277: Hybrid/Electric cars b attery current and voltage

jumpjack, 2025-02-25:
Is it possible to add functions to read battery current, voltage and  status of charge?

maybe this can help? https://www.tahoeyukonforum.com/threads/obd-scanning-realtime-hybrid-info.45961/

PowerBroker2, 2025-02-25:
Are these PIDs supported by all cars? While I'm not 100% against it, I hesitate to support non-standard PIDs. It should be easy for you to implement the functions at the sketch level - have you tried?

jumpjack, 2025-02-25:
This is the point: OBD-II has been invented to read pollution data from internal combustion engines, so there are no rules for hybrid and electric vehicles, so any manufacturer is on its own... resulting in a mess.

And I have no idea about which are the right PIDs for my car (Renault Captur PHEV), how to test them, and how to look for them. :-(

PowerBroker2, 2025-02-25:
Try looking on forums for your car and/or Torque, because I sure don't know

closed 2025-02-25 by PowerBroker2

jumpjack, 2025-02-26:
I found these  data, but can't interpret them:

https://github.com/nickn17/evDash/blob/master/src/CarRenaultZoe.cpp

jumpjack, 2025-02-26:
Also found [this](https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles.json):

# Renault Zoe Ph2 (2020-)

init: "ATE0;ATH1;ATSP7;ATS0;ATM0;ATAT1;ATFCSM1;ATCP18;"

| PID     | Name         | Expression                   | Unit  |
|---------|--------------|------------------------------|-------|
| 222006  | Odometer     | [B4:B6]                      | km    |
| 222B85  | Plugged      | B4                           |       |
| 229002  | SOC          | [B4:B5] * 0.01               | %     |
| 22300F  | PowerAvail   | [B4:B5] * 0.025              | kW    |
| 22300B  | AcPhases     | B4                           |       |
| 222218  | TempAmb      | ([B4:B5] * 0.1) - 273        | °C    |
| 222005  | Batt12v      | [B4:B5] * 0.01               | V     |
| 229003  | SOH          | [B4:B5] * 0.01               | %     |
| 2291C8  | KwhAvail     | [B4:B6] * 0.001              | kwh   |
| 229243  | KwhCharged   | [B5:B8] * 0.001              | kwh   |
| 22300D  | AcCurrent    | [B4:B5] * 0.1                | A     |
| 222B8A  | AcVoltage    | [B4:B5] * 0.5                | V     |

# Renault Zoe R110/R90
init: "ATSP6;"

| PID     | Name         | Expression     | Unit  |
|---------|--------------|----------------|-------|
| 222002  | SoC          | [B4:B5]*0.02   | %     |
| 223206  | SoH          | B4             | %     |
| 223451  | Range        | [B4:B5]        | km    |
| 222006  | Odometer     | [B4:B6]        | km    |

# Renault: Kangoo Z.E
init: "ATSP6;"

| PID     | Name | Expression        | Unit |
|---------|------|-------------------|------|
| 222002  | SoC  | ((B4*256)+B5)/50  | %    |

# Renault: Twizy
init: "ATSP6;ATST96;ATSH79B;ATFCSH79B;ATFCSD300000;ATFCSM1;ATCRA7BB;",

| PID    | Name | Expression      | Unit |
|--------|------|-----------------|------|
| 21035  | SOC  | [B29:B30]/100   | %    |

jumpjack, 2025-02-26:
[Further data](https://github.com/meatpiHQ/wican-fw/discussions/238) (with init strings for each PID):

| Name       | PID     | Expression             | Unit | Init                                                     |
|------------|---------|------------------------|------|----------------------------------------------------------|
| Odometer   | 222006  | [B4:B6]                | km   | ATSP7;ATFCSH18DADAF1;ATCRA18DAF1DA;ATSHDADAF1;ATFCSM1;   |
| Plugged    | 222B85  | B4                     |      | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| Soc        | 229002  | [B4:B5] * 0.01         | %    | ATSP7;ATSHDADBF1;ATFCSH18DADBF1;ATFCSM1;ATCRA18DAF1DB;   |
| PowerAvail | 22300F  | [B4:B5] * 0.025        | kW   | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| AcPhases   | 22300B  | B4 + 1                 |      | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| 12vBat     | 222005  | [B4:B5] * 0.01         | V    | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| AcCurrent  | 22300D  | [B4:B5] * 0.1          | A    | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| AcVoltage  | 222B8A  | [B4:B5] * 0.5          | V    | ATSP7;ATSHDADAF1;ATFCSH18DADAF1;ATFCSM1;ATCRA18DAF1DA;   |
| Soh        | 229003  | [B4:B5] * 0.01         | %    | ATSP7;ATSHDADBF1;ATFCSH18DADBF1;ATFCSM1;ATCRA18DAF1DB;   |
| CompState  | 224402  | B4                     |      | ATSP6;ATFCSH744;ATCRA764;ATSH744;ATFCSM1;                |
| Locked     | 228004  | B4                     |      | ATSP6;ATFCSH745;ATCRA765;ATSH745;ATFCSM1;                |
| CabinTemp  | 224009  | ([B4:B5] - 400) / 10   | °C   | ATSP6;ATFCSH744;ATCRA764;ATSH744;ATFCSM1;                |

ligius-, 2025-05-19:
They are non-standard and vary widely across cars. On a model that I'm working on (not Renault) the Torque definitions are different, both in terms of PIDs and formulas. The only long-term solution is to have some kind of extension module for each car make and model.