PowerBroker2/ELMduino #278: How to implement custom PIDs?

jumpjack, 2025-02-27:
Given these data:

| Name       | PID     | Expression             | Unit | Init                                                     |
|------------|---------|------------------------|------|----------------------------------------------------------|
| Soc        | 229002  | [B4:B5] * 0.01         | %    | ATSP7;ATSHDADBF1;ATFCSH18DADBF1;ATFCSM1;ATCRA18DAF1DB;   |

How do I implement in the source code the call to PID 229002?  I.E. how do I get Service and PID from number 229002?

jimwhitelaw, 2025-02-28:
Have a look at this example sketch. ELMduino will parse the PID for you. (FYI, it's Service:0x22 and PID 0x9002 above).
https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_CustomHeader/ESP32_CustomHeader.ino

Maximize0987, 2025-03-04:
> Have a look at this example sketch. ELMduino will parse the PID for you. (FYI, it's Service:0x22 and PID 0x9002 above). https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_CustomHeader/ESP32_CustomHeader.ino

So does jumpjack change myELM327.sendCommand("**2204FE**");  to     myELM327.sendCommand("**229002**");

asking because I am looking for a ford transmission temp that is   221674    **Thanks**

jimwhitelaw, 2025-03-05:
@Maximize0987, yes that’s correct. In your case you would call 
```c
myELM327.sendCommand("221674”);
```

Maximize0987, 2025-03-05:
any help on formulas?  I am hoping to include these in a sketch  would the call you stated return a and b?

ModeAndPID Name                           Min Max Unit    Formula                                 Header
221E1C     Transmission Temp              0   300 F       (((A*256)+B)*(9/8)+320)/10      0
22DD04     Inside Car Temp                0   300 F        (A*18-400)/10                            0
221674     Trans Fluid Temp               0   250 F         ((A*256)+B)/8                            0
221310     Engine Oil Temp                0   250 C        (((A*256)+B)/100)-40                 0
221624     Cyl Head Temp                                        ((A*256)+B)*(2)

Maximize0987, 2025-03-06:
> [@Maximize0987](https://github.com/Maximize0987), yes that’s correct. In your case you would call
> 
> myELM327.sendCommand("221674”);

So I think I need to check the different bytes to find the A and B values then apply the math until I find a

answer that equals my scan tool

in the example          int A = myELM327.payload[6];     

I found in another git someone was looking for 221e1c trans temp where I have the formula (((A*256)+B)*(9/8)+320)/10

their sketch used rxWord1[9]  and  rxWord1[12] in their equation

so I'll start with payload[9] and payload[12] and then try offsets from there.

CAN YOU RECOMMEND A INEXPENSIVE BLUETOOTH OBD ADAPTER THAT IS SAFE TO NOT KILL THE BATTERY WHEN PARKED????

THanks again

Maximize0987, 2025-03-06:
@jimwhitelaw
ok that is not my biggest concern  I have a obdlink mx+ that seems to be working, I just wouldn't prefer to have a $140 device set to this task if it can be done with a < $30 device.

I just did a test run with errors, and I'm not sure atm if I have correct header, and if so I am getting timeouts of 0ms which seems like with bluetooth is a very short timeout.

09:26:12.590 Connected to CP210x device
09:26:16.355 Clearing input serial buffer
09:26:16.359 Sending the following command/query: AT D
09:26:16.368 ^IReceived char: A
09:26:16.370 ^IReceived char: T
09:26:16.373 ^IReceived char: _
09:26:16.373 ^IReceived char: D
09:26:16.374 ^IReceived char: \r
09:26:16.376 ^IReceived char: O
09:26:16.378 ^IReceived char: K
09:26:16.380 ^IReceived char: \r
09:26:16.382 ^IReceived char: \r
09:26:16.383 ^IReceived char: >
09:26:16.385 Delimiter found.
09:26:16.387 All chars received: ATDOK
09:26:16.478 Clearing input serial buffer
09:26:16.481 Sending the following command/query: AT Z
09:26:16.720 ^IReceived char: A
09:26:16.722 ^IReceived char: T
09:26:16.723 ^IReceived char: _
09:26:16.725 ^IReceived char: Z
09:26:16.727 ^IReceived char: \r
09:26:17.350 ^IReceived char: \r
09:26:17.353 ^IReceived char: \r
09:26:17.353 ^IReceived char: E
09:26:17.355 ^IReceived char: L
09:26:17.357 ^IReceived char: M
09:26:17.358 ^IReceived char: 3
09:26:17.360 ^IReceived char: 2
09:26:17.362 ^IReceived char: 7
09:26:17.363 ^IReceived char: _
09:26:17.364 ^IReceived char: v
09:26:17.366 ^IReceived char: 1
09:26:17.368 ^IReceived char: .
09:26:17.370 ^IReceived char: 4
09:26:17.372 ^IReceived char: b
09:26:17.373 ^IReceived char: \r
09:26:17.375 ^IReceived char: \r
09:26:17.378 ^IReceived char: >
09:26:17.379 Delimiter found.
09:26:17.380 All chars received: ATZELM327v1.4b
09:26:17.471 Clearing input serial buffer
09:26:17.475 Sending the following command/query: AT E0
09:26:17.519 ^IReceived char: A
09:26:17.522 ^IReceived char: T
09:26:17.523 ^IReceived char: _
09:26:17.525 ^IReceived char: E
09:26:17.526 ^IReceived char: 0
09:26:17.547 ^IReceived char: \r
09:26:17.549 ^IReceived char: O
09:26:17.551 ^IReceived char: K
09:26:17.552 ^IReceived char: \r
09:26:17.554 ^IReceived char: \r
09:26:17.555 ^IReceived char: >
09:26:17.557 Delimiter found.
09:26:17.558 All chars received: ATE0OK
09:26:17.650 Clearing input serial buffer
09:26:17.653 Sending the following command/query: AT S0
09:26:17.708 ^IReceived char: O
09:26:17.710 ^IReceived char: K
09:26:17.712 ^IReceived char: \r
09:26:17.714 ^IReceived char: \r
09:26:17.715 ^IReceived char: >
09:26:17.717 Delimiter found.
09:26:17.719 All chars received: OK
[... 99 lines cut ...]
09:26:33.062 Sending the following command/query: 221E1C
09:26:35.058 Timeout detected with overflow of 0ms
09:26:35.061 Received:
09:26:35.062 ERROR: ELM_TIMEOUT
09:26:40.158 Clearing input serial buffer
09:26:40.160 Sending the following command/query: 221E1C
09:26:42.156 Timeout detected with overflow of 0ms
09:26:42.160 Received:
09:26:42.161 ERROR: ELM_TIMEOUT
09:26:47.256 Clearing input serial buffer
09:26:47.259 Sending the following command/query: 221E1C
09:26:49.255 Timeout detected with overflow of 0ms
09:26:49.259 Received:
09:26:49.260 ERROR: ELM_TIMEOUT
09:26:54.356 Clearing input serial buffer
09:26:54.358 Sending the following command/query: 221E1C
09:26:56.355 Timeout detected with overflow of 0ms
09:26:56.358 Received:
09:26:56.359 ERROR: ELM_TIMEOUT
09:27:01.455 Clearing input serial buffer
09:27:01.457 Sending the following command/query: 221E1C
09:27:03.454 Timeout detected with overflow of 0ms
09:27:03.457 Received:
09:27:03.458 ERROR: ELM_TIMEOUT
09:27:04.303 Disconnected