PowerBroker2/ELMduino #279: Issue with responseByte handling on ELM327 v3.3.1

ElettroshockNow, 2025-03-01:
Good evening, everyone,

I need to process PID 122, which returns 7 data bytes.

Here is the code I am using:

dpf = myELM327.processPID(01, 122, 1, 7, 1, 0);
if (myELM327.nb_rx_state == ELM_SUCCESS) {
    float B = myELM327.responseByte_6; // Confirmed to be B (7=A, 6=B, 5=C, 4=D, 3=E, 2=F, 1=G)
    float C = myELM327.responseByte_5; // Confirmed to be C (1=G, 2=F, 3=E, 4=D, 5=C, 6=B, 7=A)
    float dpf_calc_respons = ((B * 256) + C) / 100;
}
.............

With version 3.3.0, I was able to solve the issue by modifying the library and adding responseByte handling as follows:

responseByte_0 = response & 0xFF;
responseByte_1 = (response >> 8) & 0xFF;
responseByte_2 = (response >> 16) & 0xFF;
responseByte_3 = (response >> 24) & 0xFF;
responseByte_4 = (response >> 32) & 0xFF;
responseByte_5 = (response >> 40) & 0xFF;
responseByte_6 = (response >> 48) & 0xFF;
responseByte_7 = (response >> 56) & 0xFF;
responseByte_8 = (response >> 64) & 0xFF;
responseByte_9 = (response >> 72) & 0xFF;
responseByte_10 = (response >> 80) & 0xFF;
responseByte_11 = (response >> 88) & 0xFF;
responseByte_12 = (response >> 96) & 0xFF;
responseByte_13 = (response >> 104) & 0xFF;

However, after applying the same modification to version 3.3.1, the code no longer works.

Has anyone encountered the same issue or have any suggestions on how to fix it?

Thanks in advance! 🚀

PowerBroker2, 2025-03-01:
Can you post the debug printouts?

ElettroshockNow, 2025-03-02:
of course :)

**1) debug printouts with library 3.3.0 modified (add responseByte):**
```
Service: 1
PID: 122
Normal length query detected
Query string: 017A1
Clearing input serial buffer
Sending the following command/query: 017A1
	Received char: 0
	Received char: 0
	Received char: 9
	Received char: \r
	Received char: 0
	Received char: :
	Received char: 4
	Received char: 1
	Received char: 7
	Received char: A
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 4
	Received char: 6
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: 1
	Received char: :
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 0090:417A010046001:00000000000000
Expected response header: 417A
Single response detected
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 70
	responseByte_6: 0
	responseByte_7: 27
	responseByte_8: 0
	responseByte_9: 0
	responseByte_10: 0
	responseByte_11: 0
	responseByte_12: 0
	responseByte_13: 0
Looking for lagging zeros
Lagging zeros not found - assuming leading zeros
dpf: 0.70
[... 229 lines cut ...]
08:20:42.158 -> 	Processing hex nibble: 0
08:20:42.158 -> 	Processing hex nibble: 0
08:20:42.204 -> 	Processing hex nibble: 0
08:20:42.204 -> 	Processing hex nibble: 0
08:20:42.277 -> 	Processing hex nibble: 0
08:20:42.277 -> 	Processing hex nibble: 0
08:20:42.277 -> 	Processing hex nibble: 0
08:20:42.362 -> 	Processing hex nibble: 0
08:20:42.362 -> 	Processing hex nibble: 0
08:20:42.409 -> 	Processing hex nibble: 0
08:20:42.409 -> 	Processing hex nibble: 0
08:20:42.504 -> 	Processing hex nibble: 0
08:20:42.504 -> 	Processing hex nibble: 0
08:20:42.504 -> 	Processing hex nibble: 0
08:20:42.543 -> 64-bit response: 
08:20:42.543 -> 	responseByte_0: 0
08:20:42.543 -> 	responseByte_1: 0
08:20:42.644 -> 	responseByte_2: 0
08:20:42.644 -> 	responseByte_3: 0
08:20:42.644 -> 	responseByte_4: 0
08:20:42.697 -> 	responseByte_5: 0
08:20:42.697 -> 	responseByte_6: 0
08:20:42.697 -> 	responseByte_7: 0
08:20:42.754 -> dpf: -1.00
```

ElettroshockNow, 2025-03-02:
Log response detected at index: 18 !!!

PowerBroker2, 2025-03-02:
This is embarrassing to admit, but I'm not actually an expert on PID responses. Do you (or anyone else who may be reading this) know why some PID responses have the data after the second colon (#275) and other responses like this be after the first semicolon?

ElettroshockNow, 2025-03-03:
if a complete log using the old modified 3.3.0 library can be useful
note: without the change I received 0, same as 3.3.1.

```
12:49:01.866 -> Service: 1
12:49:01.866 -> PID: 122
12:49:01.866 -> Normal length query detected
12:49:01.866 -> Query string: 017A1
12:49:01.866 -> Clearing input serial buffer
12:49:01.866 -> Sending the following command/query: 017A1
12:49:01.969 -> 	Received char: 0
12:49:02.003 -> 	Received char: 0
12:49:02.003 -> 	Received char: 9
12:49:02.050 -> 	Received char: \r
12:49:02.050 -> 	Received char: 0
12:49:02.097 -> 	Received char: :
12:49:02.097 -> 	Received char: 4
12:49:02.139 -> 	Received char: 1
12:49:02.139 -> 	Received char: 7
12:49:02.139 -> 	Received char: A
12:49:02.177 -> 	Received char: 0
12:49:02.177 -> 	Received char: 1
12:49:02.223 -> 	Received char: 0
12:49:02.223 -> 	Received char: 0
12:49:02.270 -> 	Received char: 5
12:49:02.270 -> 	Received char: 0
12:49:02.324 -> 	Received char: 0
12:49:02.324 -> 	Received char: 0
12:49:02.324 -> 	Received char: \r
12:49:02.361 -> 	Received char: 1
12:49:02.361 -> 	Received char: :
12:49:02.400 -> 	Received char: 0
12:49:02.400 -> 	Received char: 0
12:49:02.444 -> 	Received char: 0
12:49:02.444 -> 	Received char: 0
12:49:02.478 -> 	Received char: 0
12:49:02.478 -> 	Received char: 0
12:49:02.517 -> 	Received char: 0
12:49:02.551 -> 	Received char: 0
12:49:02.551 -> 	Received char: 0
12:49:02.598 -> 	Received char: 0
12:49:02.598 -> 	Received char: 0
12:49:02.598 -> 	Received char: 0
12:49:02.640 -> 	Received char: 0
12:49:02.640 -> 	Received char: 0
12:49:02.686 -> 	Received char: \r
12:49:02.686 -> 	Received char: \r
12:49:02.734 -> 	Received char: >
12:49:02.734 -> Delimiter found.
12:49:02.734 -> All chars received: 0090:417A010050001:00000000000000
12:49:02.734 -> Expected response header: 417A
12:49:02.734 -> Single response detected
12:49:02.734 -> 64-bit response: 
12:49:02.734 -> 	responseByte_0: 0
12:49:02.734 -> 	responseByte_1: 0
12:49:02.734 -> 	responseByte_2: 0
12:49:02.734 -> 	responseByte_3: 0
12:49:02.734 -> 	responseByte_4: 0
12:49:02.734 -> 	responseByte_5: 80
12:49:02.734 -> 	responseByte_6: 0
12:49:02.734 -> 	responseByte_7: 27
12:49:02.734 -> 	responseByte_8: 0
12:49:02.734 -> 	responseByte_9: 0
12:49:02.734 -> 	responseByte_10: 0
12:49:02.734 -> 	responseByte_11: 0
12:49:02.734 -> 	responseByte_12: 0
12:49:02.734 -> 	responseByte_13: 0
12:49:02.781 -> Looking for lagging zeros
12:49:02.781 -> Lagging zeros not found - assuming leading zeros
12:49:02.781 -> dpf: 0.80
[... 271 lines cut ...]
12:49:06.100 -> 	Received char: \r
12:49:06.139 -> 	Received char: >
12:49:06.139 -> Delimiter found.
12:49:06.139 -> All chars received: 0090:417A010050001:00000000000000
12:49:06.139 -> Expected response header: 417A
12:49:06.139 -> Single response detected
12:49:06.139 -> 64-bit response: 
12:49:06.139 -> 	responseByte_0: 0
12:49:06.139 -> 	responseByte_1: 0
12:49:06.139 -> 	responseByte_2: 0
12:49:06.139 -> 	responseByte_3: 0
12:49:06.139 -> 	responseByte_4: 0
12:49:06.174 -> 	responseByte_5: 80
12:49:06.174 -> 	responseByte_6: 0
12:49:06.174 -> 	responseByte_7: 27
12:49:06.174 -> 	responseByte_8: 0
12:49:06.174 -> 	responseByte_9: 0
12:49:06.174 -> 	responseByte_10: 0
12:49:06.174 -> 	responseByte_11: 0
12:49:06.174 -> 	responseByte_12: 0
12:49:06.174 -> 	responseByte_13: 0
12:49:06.174 -> Looking for lagging zeros
12:49:06.174 -> Lagging zeros not found - assuming leading zeros
12:49:06.174 -> dpf: 0.80
```

ElettroshockNow, 2025-03-03:
NEWS 

**New, I tried another obd interface (Vlink) and reinstalled the 3.3.1, processing as below IT WORKS, but returns a WARNING, is this important?**  

```
dpf = myELM327.processPID(01, 122, 1, 7, 1, 0);
          if (myELM327.nb_rx_state == ELM_SUCCESS) {
            float B = myELM327.responseByte_2;  //confermato che è B (7=A 6=B 5=C 4=D 3=E 2=F 1=G)
            float C = myELM327.responseByte_1;  //confermato che è C (1=G 2=F 3=E 4=D 5=C 6=B 7=A)
            float dpf_calc_respons = ((B * 256) + C) / 100;
```
**1) debug printouts with library 3.3.1 and processPID(01, 122, 1, 7, 1, 0).........WORK IT**
```
18:35:53.550 -> Service: 1
18:35:53.550 -> PID: 122
18:35:53.550 -> Normal length query detected
18:35:53.550 -> Query string: 017A1
18:35:53.550 -> Clearing input serial buffer
18:35:53.550 -> Sending the following command/query: 017A1
18:35:53.550 -> 	Received char: 0
18:35:53.597 -> 	Received char: 1
18:35:53.597 -> 	Received char: 7
18:35:53.597 -> 	Received char: A
18:35:53.635 -> 	Received char: 1
18:35:53.635 -> 	Received char: \r
18:35:53.670 -> 	Received char: 0
18:35:53.720 -> 	Received char: 0
18:35:53.720 -> 	Received char: 9
18:35:53.720 -> 	Received char: \r
18:35:53.767 -> 	Received char: 0
18:35:53.767 -> 	Received char: :
18:35:53.819 -> 	Received char: 4
18:35:53.819 -> 	Received char: 1
18:35:53.819 -> 	Received char: 7
18:35:53.869 -> 	Received char: A
18:35:53.869 -> 	Received char: 0
18:35:53.920 -> 	Received char: 1
18:35:53.920 -> 	Received char: 0
18:35:53.920 -> 	Received char: 0
18:35:53.973 -> 	Received char: 6
18:35:53.973 -> 	Received char: 4
18:35:53.973 -> 	Received char: 0
18:35:54.013 -> 	Received char: 0
18:35:54.013 -> 	Received char: \r
18:35:54.051 -> 	Received char: \r
18:35:54.098 -> 	Received char: >
18:35:54.098 -> Delimiter found.
18:35:54.098 -> All chars received: 017A10090:417A01006400
18:35:54.098 -> Expected response header: 417A
18:35:54.098 -> Single response detected
18:35:54.098 -> 	Processing hex nibble: 0
18:35:54.098 -> 	Processing hex nibble: 1
18:35:54.098 -> 	Processing hex nibble: 0
18:35:54.098 -> 	Processing hex nibble: 0
18:35:54.098 -> 	Processing hex nibble: 6
18:35:54.098 -> 	Processing hex nibble: 4
18:35:54.098 -> 	Processing hex nibble: 0
18:35:54.098 -> 	Processing hex nibble: 0
18:35:54.098 -> 64-bit response: 
18:35:54.098 -> 	responseByte_0: 0
18:35:54.098 -> 	responseByte_1: 100
18:35:54.098 -> 	responseByte_2: 0
18:35:54.098 -> 	responseByte_3: 1
18:35:54.098 -> 	responseByte_4: 0
18:35:54.098 -> 	responseByte_5: 0
18:35:54.098 -> 	responseByte_6: 0
18:35:54.136 -> 	responseByte_7: 0
18:35:54.136 -> WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
18:35:54.136 -> dpf: 1.00
```

I also ran this test just out of curiosity

**2) debug printouts with library 3.3.1 and processPID(01, 122, 2, 7, 1, 0).... NOT WORK IT**
```
18:39:23.596 -> Service: 1
18:39:23.596 -> PID: 122
18:39:23.596 -> Normal length query detected
18:39:23.596 -> Query string: 017A2
18:39:23.650 -> Clearing input serial buffer
18:39:23.650 -> Sending the following command/query: 017A2
18:39:23.650 -> 	Received char: 0
18:39:23.650 -> 	Received char: 1
18:39:23.691 -> 	Received char: 7
18:39:23.691 -> 	Received char: A
18:39:23.729 -> 	Received char: 2
18:39:23.729 -> 	Received char: \r
18:39:23.765 -> 	Received char: 0
18:39:23.765 -> 	Received char: 0
18:39:23.812 -> 	Received char: 9
18:39:23.812 -> 	Received char: \r
18:39:23.856 -> 	Received char: 0
18:39:23.856 -> 	Received char: :
18:39:23.895 -> 	Received char: 4
18:39:23.895 -> 	Received char: 1
18:39:23.931 -> 	Received char: 7
18:39:23.931 -> 	Received char: A
18:39:23.966 -> 	Received char: 0
18:39:23.966 -> 	Received char: 1
18:39:24.013 -> 	Received char: 0
18:39:24.013 -> 	Received char: 0
18:39:24.063 -> 	Received char: 3
18:39:24.063 -> 	Received char: 2
18:39:24.063 -> 	Received char: 0
18:39:24.110 -> 	Received char: 0
18:39:24.110 -> 	Received char: \r
18:39:24.151 -> 	Received char: 1
18:39:24.151 -> 	Received char: :
18:39:24.198 -> 	Received char: 0
18:39:24.198 -> 	Received char: 0
18:39:24.251 -> 	Received char: 0
18:39:24.251 -> 	Received char: 0
18:39:24.251 -> 	Received char: 0
18:39:24.298 -> 	Received char: 0
18:39:24.298 -> 	Received char: 0
18:39:24.351 -> 	Received char: 0
18:39:24.351 -> 	Received char: 0
18:39:24.351 -> 	Received char: 0
18:39:24.390 -> 	Received char: 0
18:39:24.390 -> 	Received char: 0
18:39:24.429 -> 	Received char: 0
18:39:24.429 -> 	Received char: 0
18:39:24.470 -> 	Received char: \r
18:39:24.470 -> 	Received char: \r
18:39:24.518 -> 	Received char: >
18:39:24.518 -> Delimiter found.
18:39:24.518 -> All chars received: 017A20090:417A010032001:00000000000000
18:39:24.518 -> Expected response header: 417A
18:39:24.518 -> Log response detected at index: 23
18:39:24.518 -> Single response detected
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.518 -> 	Processing hex nibble: 0
18:39:24.556 -> 	Processing hex nibble: 0
18:39:24.556 -> 	Processing hex nibble: 0
18:39:24.556 -> 	Processing hex nibble: 0
18:39:24.556 -> 	Processing hex nibble: 0
18:39:24.556 -> 64-bit response: 
18:39:24.556 -> 	responseByte_0: 0
18:39:24.556 -> 	responseByte_1: 0
18:39:24.556 -> 	responseByte_2: 0
18:39:24.556 -> 	responseByte_3: 0
18:39:24.556 -> 	responseByte_4: 0
18:39:24.556 -> 	responseByte_5: 0
18:39:24.556 -> 	responseByte_6: 0
18:39:24.556 -> 	responseByte_7: 0
18:39:24.556 -> dpf: -1.00
```

PowerBroker2, 2025-03-04:
Same as #280 - will be fixed in 3.3.2. Will tag out today.

closed 2025-03-04 by PowerBroker2