PowerBroker2/ELMduino #280: Debug help

carloda, 2025-03-02:
Hello everyone, I have a problem that is probably due to an error on my part.

This is my code, and everything seems to be working correctly, but the problem is that, in addition to the RPM value, many other pieces of information are printed on the serial monitor that I don't need. I probably haven’t managed to disable the debug. Could you tell me how to solve this problem? Thanks

```
#include "BluetoothSerial.h"
#include "ELMduino.h"

const bool DEBUG        = false;
const int  TIMEOUT      = 2000;
const bool HALT_ON_FAIL = false;

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint8_t address[6] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xBA};  // MAC address del tuo ELM327

uint32_t rpm = 0;

bool debug = false;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  // Inizializzazione Bluetooth
  ELM_PORT.begin("ArduHUD", true);
  
  // Connessione tramite MAC address
  if (!ELM_PORT.connect(address))  // Usa il MAC address per la connessione
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);  // Se non riesci a connetterti, fermati qui
  }

  // Inizializzazione dell'ELM327
  if (!myELM327.begin(ELM_PORT, true, 2000))  // Timeout a 2000ms
  {
    DEBUG_PORT.println("Couldn't initialize ELM327");
    while (1);  // Se non riesci a inizializzare, fermati qui
  }

  DEBUG_PORT.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    //rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); //Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    //myELM327.printError();

  delay(100);  // Aggiungi un ritardo per ridurre la frequenza di lettura
}
```

carloda, 2025-03-02:
Fixed =)

carloda, 2025-03-02:
I solved it by setting "!myELM327.begin(ELM_PORT, false, 2000))" but now I can't solve a similar problem, they still give me other information that I don't need:

```
22.00
        Processing hex nibble: 3
        Processing hex nibble: E
```

How do I unbait them? Thank you

PowerBroker2, 2025-03-02:
Thank you for bringing this up, I had a stupid mistake that is now fixed in 3.3.2

commit 82c7d8c8 (2025-03-02): Fix #280
closed 2025-03-02 by PowerBroker2