PowerBroker2/ELMduino #283: Watchdog Triggered Cheap Yellow Display

brickhockey3, 2025-03-13:
`#include <lvgl.h>
#include <TFT_eSPI.h>
#include <XPT2046_Touchscreen.h>
#include <BluetoothSerial.h>
#include <ELMduino.h>
#include <atomic>

#define SCREEN_WIDTH 240
#define SCREEN_HEIGHT 320
#define XPT2046_IRQ 36
#define XPT2046_MOSI 32
#define XPT2046_MISO 39
#define XPT2046_CLK 25
#define XPT2046_CS 33

#define LCD_BACK_LIGHT_PIN 21
#define LEDC_CHANNEL_0     0
#define LEDC_TIMER_12_BIT  12
#define LEDC_BASE_FREQ     5000

SPIClass touchscreenSPI(VSPI);
XPT2046_Touchscreen touchscreen(XPT2046_CS, XPT2046_IRQ);
TFT_eSPI tft = TFT_eSPI();
uint32_t draw_buf[SCREEN_WIDTH * SCREEN_HEIGHT / 10];

BluetoothSerial SerialBT;
ELM327 obd;
std::atomic<uint32_t> rpm{0};
std::atomic<uint32_t> coolant_temp{0};
std::atomic<uint32_t> voltage{0};
std::atomic<uint32_t> tpsval{0};
uint32_t last_displayed_rpm = -1;
uint32_t last_displayed_temp = -1;
uint32_t last_displayed_voltage = -1;
uint32_t last_displayed_tpsval = -1;

SemaphoreHandle_t lvgl_mutex;
TaskHandle_t lvgl_task_handle;
TaskHandle_t obd_task_handle;
TaskHandle_t bt_icon_task_handle;

static lv_obj_t *status_label;
static lv_obj_t *rpm_label;
static lv_obj_t *rpm_val;
static lv_obj_t *coolant_temp_label;
static lv_obj_t *coolant_temp_val;
static lv_obj_t *voltage_label;
static lv_obj_t *voltage_unit_label;
static lv_obj_t *voltage_val;
static lv_obj_t *TPS_label;
static lv_obj_t *TPS_unit_label;
static lv_obj_t *TPS_val;
static lv_obj_t *bt_icon_label;
static lv_obj_t *toggle_btn;
static lv_obj_t *settings_btn;
static lv_obj_t *settings_scr;
static lv_obj_t *main_scr;
static lv_obj_t *brightness_slider;

bool use_celsius = false;
uint8_t current_brightness = 128;

void lvgl_task(void *parameters);
void obd_task(void *parameters);
void bt_icon_task(void *parameters);
void show_main_screen();
void show_settings_screen();

void ledcAnalogWrite(uint8_t channel, uint32_t value, uint32_t valueMax = 255) {
  // calculate duty, 4095 from 2 ^ 12 - 1
[... 452 lines cut ...]
	Received char: O
	Received char: _
	Received char: C
	Received char: O
Received char: N
	Received char: N
	Received char: E
	Received char: C
	Received char: T
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: SEARCHING...UNABLETOCONNECT
ELM responded with error "UNABLE TO CONNECT"
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK`

jimwhitelaw, 2025-03-13:
Since you know the protocol you want to connect with, try specifying that in _obd.begin()_. That will avoid the delay incurred with searching for the correct protocol.

brickhockey3, 2025-03-13:
Well my car uses OBDII and my truck uses SAEJ1850, it would be great to make the device universal in also if I wanted to give them to friends.

brickhockey3, 2025-03-13:
> Since you know the protocol you want to connect with, try specifying that in _obd.begin()_. That will avoid the delay incurred with searching for the correct protocol.

Well my car uses OBDII and my truck uses SAEJ1850, it would be great to make the device universal in also if I wanted to give them to friends.

jimwhitelaw, 2025-03-14:
In both cases, the failures are occurring at the initialization step, where ELM327 is doing the auto search for the correct protocol. The second failure isn't a timeout per se, it's a notice that the ELM327 device simply can't find a compatible ECU/protocol. It's possible the first example would return the same, if it weren't killed by the watchdog first. Are you certain that your ELM device works with the vehicles you're working with? Does it work with other apps or one of the simple sketches like examples/ESP32_Bluetooth_Serial?

brickhockey3, 2025-03-14:
> In both cases, the failures are occurring at the initialization step, where ELM327 is doing the auto search for the correct protocol. The second failure isn't a timeout per se, it's a notice that [...]

     So ESP_Bluetooth_multiple_pids  connects to both OBDII and SAEj1850 without problems. I will include the serial output from that now with my SAEj1850 since that is what is causing problems for me. 
```
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: A
	Received char: T
	Received char: _
	Received char: D
	Received char: \r
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATDOK
Clearing input serial buffer
Sending the following command/query: AT Z
	Received char: A
	Received char: T
	Received char: _
	Received char: Z
	Received char: \r
	Received char: \r
	Received char: \r
	Received char: E
	Received char: L
	Received char: M
	Received char: 3
	Received char: 2
	Received char: 7
	Received char: _
	Received char: v
	Received char: 1
	Received char: .
	Received char: 5
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char: _
	Received char: E
	Received char: 0
	Received char: \r
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
[... 178 lines cut ...]
	Received char: 1
	Received char: 0
	Received char: D
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 410D00
Expected response header: 410D
Single response detected
	Processing hex nibble: 0
	Processing hex nibble: 0
64-bit response: 
	responseByte_0: 0
	responseByte_1: 0
	responseByte_2: 0
	responseByte_3: 0
	responseByte_4: 0
	responseByte_5: 0
	responseByte_6: 0
	responseByte_7: 0
mph: 0.00
```

jimwhitelaw, 2025-03-14:
So after a couple failed attempts to connect, it does finally sync up and return some results. The rpm and mph values are 0 presumably because the engine is off and vehicle is not moving. The way the protocol search works in an auto search is that after receiving the SP0 command, the ELM327 does not actually initiate the search immediately; instead, it initiates the search on the next query sent. The library tries to manage this for you and even extends the default timeout during this search. In this case, you get an "Unable to connect" message follow by a valid connection a few queries later. I honestly don't know what's happening there. Do you have more than one device you can test with?

brickhockey3, 2025-03-14:
> So after a couple failed attempts to connect, it does finally sync up and return some results. The rpm and mph values are 0 presumably because the engine is off and vehicle is not moving. The way [...]

     So the only other ELM device I have is WIFI. Are you able to tell me how to extend the watchdog timer for that function so it has time to fail, then default to the other protocols ?  If I am comprehending the CPP in the library correctly it seems that when the connection protocol is not specified it will call an AT Z to reset an AT E0 to disable echo, then a AT SP0, after that it sends 0100 to check for supported PIDS, if that returns a timeout it will move onto the next protocol and then send the 0100 again. That is provided I am comprehending the CPP file and how it operates in a basic sketch without protocol specification.

brickhockey3, 2025-03-14:
> So after a couple failed attempts to connect, it does finally sync up and return some results. The rpm and mph values are 0 presumably because the engine is off and vehicle is not moving. The way [...]

     The only other ELM device I have is WIFI. Are you able to tell me how to extend the watchdog timer for that function so it has time to fail, then default to the other protocols ?  If I am comprehending the CPP in the library correctly it seems that when the connection protocol is not specified it will call an AT Z to reset an AT E0 to disable echo, then a AT SP0, after that it sends 0100 to check for supported PIDS, if that returns a timeout it will move onto the next protocol and then send the 0100 again. That is provided I am comprehending the CPP file and how it operates in a basic sketch without protocol specification.

jimwhitelaw, 2025-03-15:
https://github.com/PowerBroker2/ELMduino/blob/286023d3fd0b0fd525ea7669fa64dc0465645854/src/ELMduino.cpp#L122-L125
https://github.com/PowerBroker2/ELMduino/blob/286023d3fd0b0fd525ea7669fa64dc0465645854/src/ELMduino.cpp#L123-L125