PowerBroker2/ELMduino #286: Query to fast ?

FrittenFett96, 2025-03-21:
Hi, 
whenever I run a query, the car's OBD2 bus is somehow disrupted. The battery and service symbols light up on the car's onboard computer. The idle speed also fluctuates. It seems to me as if too many queries are being run in too short a time. I've read that you can only run a few queries per second, otherwise the OBD2 bus will be affected. I tried both the blocking and the non-blocking variant (example code) and also tested a delay in the loop with the same result. 

Car:
Opel (Vauxhall) Corsa D

Hardware:
Esp32-S3
OBD2 WiFi Adapter (works fine with android App)

Since I haven't found a similar case involving the Elmduino library, I think the library's time management is well-regulated. Does anyone have an idea what else could be the cause?

jimwhitelaw, 2025-03-21:
The ELMduino library doesn't enforce any limit on queries/sec to the adapter. I seem to recall the OBD2 spec limit is 20 queries/sec. In cases where I need to rate-limit, I typically use a timer of 200msec between queries to be safe. I've never done throughput tests, but I think that Wifi is slower than BT. Since adapters vary so much in quality, testing with another adapter is a good idea.

FrittenFett96, 2025-03-21:
Can you recommend Bluetooth ble adapters that will definitely work?

Thanks for the help.

jimwhitelaw, 2025-03-22:
This is good advice:
https://www.carscanner.info/choosing-obdii-adapter/

I’ve had success with the Vgate and Veepeak budget adapters. I believe the OBDlink MX+ is the current gold standard, and priced accordingly.

FrittenFett96, 2025-03-23:
That report is great, thanks. I actually used a cheap adapter, which is now going back. I ordered the Vgate iCar Pro 2S and will report back on the results.