PowerBroker2/ELMduino #287: Totalbytes settable .

ElettroshockNow, 2025-03-23:
Hello everyone, 
the request comes from a problem I found in the pid request in service mode 22.

Specifically the PID 22 D525, but essentially on all pids in mode 22 (OPEL,PEUGEOT) with OBD Vgate Icar Pro.

In this mode, in the response to the pid there is no value consistent with the totalbyte (line="203"--totalbyte=6!!!), therefore in the next phase the "Response data" is truncated in advance (see log below) ... in reality the totalbyte = 6 comes from an overflow.

Currently I have solved the problem by forcing the totalbyte, but it would be nice to force it manually in case of need.

Hello and thanks for everything
Luca

```
14:21:44.398 Service: 34
14:21:44.398 PID: 54565
14:21:44.398 Long query detected
14:21:44.398 Query string: 22D5251
14:21:44.403 Clearing input serial buffer
14:21:44.403 Sending the following command/query: 22D5251
14:21:44.479 	Received char: 2
14:21:44.499 	Received char: 2
14:21:44.519 	Received char: D
14:21:44.539 	Received char: 5
14:21:44.559 	Received char: 2
14:21:44.579 	Received char: 5
14:21:44.599 	Received char: 1
14:21:44.619 	Received char: \r
14:21:44.639 	Received char: 2
14:21:44.659 	Received char: 0
14:21:44.679 	Received char: 3
14:21:44.699 	Received char: \r
14:21:44.719 	Received char: 0
14:21:44.739 	Received char: :
14:21:44.759 	Received char: 6
14:21:44.779 	Received char: 2
14:21:44.799 	Received char: D
14:21:44.819 	Received char: 5
14:21:44.839 	Received char: 2
14:21:44.859 	Received char: 5
14:21:44.879 	Received char: A
14:21:44.899 	Received char: 2
14:21:44.919 	Received char: 4
14:21:44.940 	Received char: 1
14:21:44.959 	Received char: 0
14:21:44.979 	Received char: 0
14:21:44.999 	Received char: \r
14:21:45.019 	Received char: \r
14:21:45.043 	Received char: >
14:21:45.043 Delimiter found.
14:21:45.043 All chars received: 22D5251
203
0:62D525A24100

14:21:45.048 Found line in response: 22D5251
14:21:45.048 Found header in response line: 22D5251
14:21:45.054 Found line in response: 203
14:21:45.054 totalBytes = 6
14:21:45.059 Found line in response: 0:62D525A24100
14:21:45.059 Response data: 62D525A24100
14:21:45.065 Parsed multiline response: 62D525
14:21:45.065 Expected response header: 62D525
14:21:45.070 Single response detected
14:21:45.070 64-bit response: 
14:21:45.076 	responseByte_0: 0
14:21:45.076 	responseByte_1: 0
14:21:45.076 	responseByte_2: 0
14:21:45.081 	responseByte_3: 0
14:21:45.081 	responseByte_4: 0
14:21:45.081 	responseByte_5: 0
14:21:45.081 	responseByte_6: 0
14:21:45.087 	responseByte_7: 0
14:21:45.087 WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
14:21:45.097 distanza_rige_read: 0.00
```

**Library mod**
```
if (strlen(line) > 0) {
                    totalBytes = strtol(line, NULL, 16) * 2;
					if(totalBytes == 6) totalBytes = 10; //mod
                    if (debugMode) {
                        Serial.print(F("totalBytes = "));
                        Serial.println(totalBytes);
                    }
```
 **After mod** 
```
17:00:51.953 Service: 34
17:00:51.953 PID: 54565
17:00:51.953 Long query detected
17:00:51.953 Query string: 22D5251
17:00:51.959 Clearing input serial buffer
17:00:51.959 Sending the following command/query: 22D5251
17:00:52.035 	Received char: 2
17:00:52.055 	Received char: 2
17:00:52.074 	Received char: D
17:00:52.094 	Received char: 5
17:00:52.114 	Received char: 2
17:00:52.135 	Received char: 5
17:00:52.154 	Received char: 1
17:00:52.175 	Received char: \r
17:00:52.194 	Received char: 2
17:00:52.215 	Received char: 0
17:00:52.234 	Received char: 3
17:00:52.254 	Received char: \r
17:00:52.275 	Received char: 0
17:00:52.294 	Received char: :
17:00:52.315 	Received char: 6
17:00:52.334 	Received char: 2
17:00:52.354 	Received char: D
17:00:52.374 	Received char: 5
17:00:52.394 	Received char: 2
17:00:52.414 	Received char: 5
17:00:52.435 	Received char: A
17:00:52.454 	Received char: 2
17:00:52.474 	Received char: 4
17:00:52.495 	Received char: 1
17:00:52.514 	Received char: 0
17:00:52.534 	Received char: 0
17:00:52.554 	Received char: \r
17:00:52.574 	Received char: \r
17:00:52.598 	Received char: >
17:00:52.598 Delimiter found.
17:00:52.598 All chars received: 22D5251
203
0:62D525A24100

17:00:52.603 Found line in response: 22D5251
17:00:52.603 Found header in response line: 22D5251
17:00:52.609 Found line in response: 203
17:00:52.609 totalBytes = 10
17:00:52.614 Found line in response: 0:62D525A24100
17:00:52.614 Response data: 62D525A24100
17:00:52.620 Parsed multiline response: 62D525A241
17:00:52.620 Expected response header: 62D525
17:00:52.625 Single response detected
17:00:52.625 	Processing hex nibble: A
17:00:52.631 	Processing hex nibble: 2
17:00:52.631 	Processing hex nibble: 4
17:00:52.636 	Processing hex nibble: 1
17:00:52.636 64-bit response: 
17:00:52.636 	responseByte_0: 65
17:00:52.642 	responseByte_1: 162
17:00:52.642 	responseByte_2: 0
17:00:52.642 	responseByte_3: 0
17:00:52.648 	responseByte_4: 0
17:00:52.648 	responseByte_5: 0
17:00:52.648 	responseByte_6: 0
17:00:52.653 	responseByte_7: 0
17:00:52.653 distanza_rige_read: 1050.00
```

jimwhitelaw, 2025-03-25:
@ElettroshockNow 
OK, I see what's happening there. That's odd that the device is returning "203" for the value of the response length. I think what you're wanting to have is a means to specify beforehand how many bytes you're expecting in the multi-line response and to ignore the value that comes back in the response itself, is that correct?

ElettroshockNow, 2025-03-25:
@jimwhitelaw 
Hi,
yes exactly.
The request for a manual setting is because I don't seem to see delimiters that can be used with absolute certainty 22 XXXX **203** AABBZZ **00**.
Since it is a perhaps unique situation, but that could recur, the solution of the manual Totalbytes without having to touch your library would be more than satisfactory.

I'll do some further tests to confirm that strange "203" value ... I have a suspicion about the connection via BLE.

For now forcing it to 10 bytes everything works fine, but I would like to avoid modifying your work of art in the next releases.

Thanks for everything
Luca

```
10:12:34.219 Sending the following command/query: 22D5261
10:12:34.859 All chars received: 22D5261
203
0:62D526380400
----------------------------------------------------------------
10:12:40.925 Sending the following command/query: 22D5191
10:12:41.566 All chars received: 22D5191
203
0:62D519161700
----------------------------------------------------------------
10:12:47.630 Sending the following command/query: 22D5251
10:12:48.270 All chars received: 22D5251
203
0:62D525A24100
----------------------------------------------------------------
10:12:54.335 Sending the following command/query: 22D52F1
10:12:54.975 All chars received: 22D52F1
203
0:62D52F240100
----------------------------------------------------------------
10:13:01.025 Sending the following command/query: 22D5211
10:13:01.654 All chars received: 22D5211
203
0:62D521000100
```